supermerill/SuperSlicer #320: Add Linear Advance / Pressure Advance Calibration

yschroeder, 2020-06-27:
### Version
2.2.52

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro with Klipper

Marlin has a generator on their website for calibrating linear advance:
https://marlinfw.org/tools/lin_advance/k-factor.html

It would be nice if some version of this would be integrated into the calibration menu. This would avoid filling in all the values into the web version. Most of them are already known to SuperSlicer.

This calibration pattern also works well to calibrate `Pressure Advance` as it is called in Klipper. Just replace the GCode that sets the K factor with the corresponding Klipper GCode. I am aware that Klipper has a different way of calibrating `Pressure Advance`. However, their version wastes a lost of plastic as a huge calibration object is printed.

Argolein, 2020-07-04:
Different firmware would require a different handling of gcode and testing range.

Marlin's linear advance 1.0 and 1.5 differ in value but use the same gcode: M900 [K<kfactor>] [L<kfactor>] [S<slot>] [T<index>]
For RepRap: M572 D0 <value>
For Klipper I've found but I can't validate if it's correct as I don't have a printer with running Klipper: SET_PRESSURE_ADVANCE [EXTRUDER=<config_name>] [ADVANCE=<pressure_advance>] [SMOOTH_TIME=<pressure_advance_smooth_time>]: Set pressure advance parameters. If EXTRUDER is not specified, it defaults to the active extruder.

I assume this would need much coding to support every printer FW.
An easier way would be a generator where you manually enter the corresponding gcode for linear advance/pressure advance and the range to test.

supermerill, 2020-09-24:
sort of duplicate of #282

nanoplane, 2021-03-15:
So, for reprap gCode, I think the implementation has a flaw..  The M572 command applies to the "driving" extruders (D1-x).  With reprap, you can create "virtual" extruders (as with Martin), and when M572 commands are sent to these extruders, it barfs... because it's not a valid drive.....

I think since there's no way to tell if an extruder is virtual or not.. (at least in the current implementation).  We probably need a way to shut this off so we don't send commands that create errors with no way to fix it...

arcticlinux, 2022-06-08:
I make a quick attempt at replicating the Marlin test print wizard a few years ago, which I believe was written by Sineos if I recall correctly, which is what I based my repository on.

https://github.com/B73Labs/pressure-advance-js