supermerill/SuperSlicer #321: Feature Request: A better way to do countersunk holes on the bottom

LaserBee, 2020-06-27:
This is something I run into quite often. I put a lot of countersunk bolt holes in my prints. If they are on the bottom of the print, they slice like this:

https://imgur.com/7XheFfq

As you can see, the dark blue bits cause problems: it prints them first when they aren't attached to anything. So the "roof" of the countersunk hole gets really messy. So here's what I'm thinking for a solution, if possible. The first bridging layer would go straight across the large hole, missing the few lines in the middle. The next layer up would do them across, also missing the middle few, This makes a square hole, but then the next layer up can be round, having the square below it to rest on. Does this make sense? 

Like this!

https://imgur.com/fCNSdSu

Anyway, it might not be possible, but it would be useful!

yschroeder, 2020-06-27:
This would also be useful for nuts put into the part from the bottom. I regularly create geometry in the part to trick the slicer into slicing it the way you propose.

supermerill, 2020-06-27:
Use one of the "no perimeters on bridge areas" option (perimeters/shell tab, under Advanced section)

yschroeder, 2020-06-28:
I am aware that this bridge mode exists, but the geometry proposed here is a far superior solution. Please consider leaving this issue open as a reminder for the future.

supermerill, 2020-06-28:
I know but the difficulty to implement it isn't the same. There's a reason i didn't do it from the get-go.

LaserBee, 2020-07-02:
Hmm, none of those options are perfect, but I will have a play around and see if any of them work well enough. Thank you for being so responsive! 

Unrelated: I love the calibration section. That's going to be super useful to me. 

I know i've asked before, but do you have a donations page/gofundme/patreon? I'd love to support this project!

supermerill, 2020-07-02:
> do you have a donations page/gofundme/patreon? I'd love to support this project!

see "sponsor" on top of the page

LaserBee, 2020-07-05:
Done! Thank you for your hard work!