supermerill/SuperSlicer #322: Total filament used

robthide37, 2020-06-28:
### Version
SuperSlicer
Version:   2.2.52
Build:     SuperSlicer_2.2.52+UNKNOWN

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,068MB
OpenGL installation
GL version:   4.5.13521 Compatibility Profile Context 24.20.11001.5003
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon (TM) RX 580
GLSL version: 4.50

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### 3D printer brand / version + firmware version Klipper custom core xy
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_ Total filament used and layers are reading as 0 at the end of the file so dw2c reads 100 percent the whole time

supermerill, 2020-06-28:
#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

robthide37, 2020-06-28:
It does it on any file i make i had to compress to a zip. I was gonna change extension but didn't know if that would cause issue
[square_tower.zip](https://github.com/supermerill/SuperSlicer/files/4842503/square_tower.zip)

robthide37, 2020-06-28:
It may just be when you select klipper as gcode flavor that it doesn't list the total layers and filament used. I will switch to marlin and check

supermerill, 2020-06-28:
[image]

robthide37, 2020-06-28:
no you are misunderstanding open the gcode file is missing the total layers and whatever other info that duet web control and octoprint use for the time calculations
[image]

robthide37, 2020-06-28:
[image]
i prob explained it wrong the slicer has the right info but the gcode file is missing it

supermerill, 2020-06-28:
that's most probably because they didn't update for the rename.
if you change the first line from '; generated by SuperSlicer' by '; generated by PrusaSlicer', it should work, no?

n8bot, 2020-06-28:
> that's most probably because they didn't update yet for the rename from slic3r++ to superslicer.
> if you change the first line from '; generated by SuperSlicer' by '; generated by PrusaSlicer', it works, no?

RRF does not use the name of the slicer to interpret other parts of the GCode file. The name of the slicer is merely passed-on as one of the detected strings.

Object height, layer height, first layer height, and total filament length are detected by various methods, but they should capture the PrusaSlicer method of reporting filament length, regardless of the name of the slicer.

As long as the gcode file contains a line like this: ```; filament used [mm] = 10995.1``` the filament length should be detected properly.

Here is the file in RRF that contains the code that detects the various strings. It's a bit hard to understand at first, but it's pretty well-commented: https://github.com/Duet3D/RepRapFirmware/blob/dev/src/Storage/FileInfoParser.cpp

robthide37, 2020-06-29:
> > that's most probably because they didn't update yet for the rename from slic3r++ to superslicer.
> > if you change the first line from '; generated by SuperSlicer' by '; generated by PrusaSlicer', it works, no?
> 
> ~RRF does not use the name of the slicer to interpret other parts of the GCode file. The name of the slicer is merely passed-on as one of the detected strings.~
> 
> ~Object height, layer height, first layer height, and total filament length are detected by various methods, but they should capture the PrusaSlicer method of reporting filament length, regardless of the name of the slicer.~
> 
> ~As long as the gcode file contains a line like this: `; filament used [mm] = 10995.1` the filament length should be detected properly.~
> 
> ~Here is the file in RRF that contains the code that detects the various strings. It's a bit hard to understand at first, but it's pretty well-commented: https://github.com/Duet3D/RepRapFirmware/blob/dev/src/Storage/FileInfoParser.cpp#L651~
> 
> Edit: Ignore this. I now notice that the user is using Klipper firmware. Perhaps Klipper does not pass this info onto DWC2. DWC2 itself does not parse the info, it usually expects to receive it from RRF.

I will try supermerill suggestion and klipper isn't passing the info because i am using a virtual sd card that dw2c is privy to.

robthide37, 2020-06-29:
better explained is that dw2c is handling the file i am sending to dw2c it not opening file command line with klipper.

robthide37, 2020-06-29:
Yes it worked i am looking through to edit the dw2c as obv its the prob not superslicer thankyou

robthide37, 2020-06-29:
i fixed the dwc2.py file because the newest klipper commits broke it a few times so i consolidated all the fixes on the diff forks and also added the ability to parse the new naming if anyone needs here ya go
[web_dwc2.py.txt](https://github.com/supermerill/SuperSlicer/files/4845370/web_dwc2.py.txt)

robthide37, 2020-06-29:
i opened just cause i wanted you to take look and make sure this seems like right way to fix ill close issue either way but if not i will work on diff solution. I just know alot of times you fix one thing and break another

supermerill, 2020-06-29:
I'm not a dev of this software, if you want to someone to review these changes, it's on their github.
Here, i can alter the gcode output if something is wrong or missing, but nothing else.

robthide37, 2020-06-29:
> I'm not a dev of this software, if you want to someone to review these changes, it's on their github.
> Here, i can alter the gcode output if something is wrong or missing, but nothing else.

is their anyway you can switch the gcode output back or tell me how to. My attempt helps the browser portion but it still did the same thing as far as the regex calculations so it was a fail for complete fix

robthide37, 2020-06-29:
either way thankyou for your time if you have a patron or anything send me link as ill make a donation for your time

supermerill, 2020-06-29:
> switch the gcode output back
It's hardcoded, the code that generate it is utils.cpp:607, and it use the Define that contain the software name.
You can recompile it , changing the name of the software to something that works, or create a script that modify the first line and add it to the post-process script (output page)

> if you have a patron
see "support" at the top of the page.

robthide37, 2020-06-30:
Changing the name worked . my edit of the dwc2 didn't i am not to sure how to do a script for slicer but i will be attempting to do that as i don't want to be stuck recompiling every update. Thankyou for your time if anyone reading this wants to make script to change superslicer to prusaslicer in gcode lmk thanks

closed 2020-06-30 by robthide37