supermerill/SuperSlicer #323: Auto speed perimeters not adhering to % based values

kizza42, 2020-06-28:
### Version
2.2.52

### Operating system type + version
Win10 2004

### 3D printer brand / version + firmware version (if known)
Tevo Little Monster

### Behavior
Maybe I'm misunderstanding Auto speed but I always assumed it gave a max speed and then every component under it would run at a arbitrarily slower speed due to detail or adhere to the overrides provided. In this case I expect my outer perimeters to go at 30% of whatever I set my Max speed to but it seems to ignore the % value. When entering a absolute value the override is enforced.
[image]

#### Project File (.3MF) where problem occurs
[20200628 - Fanatec Keyboard Holder.zip](https://github.com/supermerill/SuperSlicer/files/4841971/20200628.-.Fanatec.Keyboard.Holder.zip)
Possible to allow uploading 3mf direct?

supermerill, 2020-06-28:
you have to set the field to 0 to use the autospeed for it

kizza42, 2020-06-28:
I understand that but when filling the fields with % values they're not enforced at all.

closed 2021-06-04 by supermerill