supermerill/SuperSlicer #324: [FR] ARC Support

thierryzoller, 2020-06-28:
Overall explanation:  https://bernhardkubicek.soup.io/post/191097625/If-you-have-a-path-of-segments
Alpha Slic3r code :https://github.com/TheThirdOne/Slic3r/commits/arc-fitting

yschroeder, 2020-06-28:
Some more info to consider:
https://community.octoprint.org/t/new-plugin-anti-stutter-need-testers/18077 

There is a plugin for octoprint in the making to convert g1 to g2/g3 post-slicing. 

SuperSlicer won't be able to do much more than this when it runs off STL files. Only when slicing STEP files it can do better.

supermerill, 2020-06-28:
problem is with step-import, is that i need to do offset, difference, union, intersection on polygons. Currently, using clipper, it's possible for a vector of points. If you add bezier curve, it may need a rewrite, be slower, etc...

n8bot, 2020-06-28:
Let's also keep in mind G5 bezier curves. Maybe G5 bezier curves do not need to be implemented at the same time, but they should be kept in mind so that any STEP file or G2/3 arc code doesn't need to be completely refactored to accommodate future implementation of G5 bezier curves.

thierryzoller, 2020-06-28:
One idea also would be (if that make sense) to use ARC support for Infill (example, gyroid)

thierryzoller, 2020-06-28:
> SuperSlicer won't be able to do much more than this when it runs off STL files. Only when slicing STEP files it can do better.
Superslicer could use this for infill, thus not only postprocessing.

thierryzoller, 2020-06-28:
Personally I am not interested in ARC support because of judder, the recent changes to marlin allowing JD but disabling certain functions solved the issue. I am not getting really round shapes, and there is tons and tons of overhead printing straight lines. Maybe if a slicer could also support another fileformat than stl that would make arc more powerfull as well.

supermerill, 2020-06-28:
the gyroid is computed by resolving an equation for each point (step).
outputing curve/bezier for gyroid means redoing it entirely, from the math formula (unless you use the post-processing script mentioned above).

The maybe only "not extremely hard" possibility is for archimedean curves.... not worth it.

for doing offset with bezier... https://stackoverflow.com/questions/4148831/how-to-offset-a-cubic-bezier-curve seems hard or slow

ciro1138, 2020-08-10:
I am curious if arc support could be added by calculating the arc based on the point sections or take the inside curve of the line sections. I know 3d systems did something like this with there sla software and it made some improvements.

ETE-Design, 2020-11-10:
@supermerill Would be great with ARC support, what I know of no slicer support that feature atm.... So would be great not to be locked to OctoPrint + ARC Welder for now...

yschroeder, 2020-11-10:
Arc welder can be used stand alone by compiling an executable. You can use it to post process your file. The octoprint plugin is just a gui for that executable.

ETE-Design, 2020-11-10:
@yschroeder Is there an easy way to "Integrate" it in any slicer then? Cura / SuperSlicer ect.?

supermerill, 2020-11-10:
And the slicer support will be the same. It's just more convenient.

you can integrate yourself by putting the command in the post-processing field. I don't know the details.

uis246, 2022-01-31:
Maybe add STEP and CSG support?

supermerill, 2022-02-01:
@uis246 please... did you read the conversation, or you're a bot?

uis246, 2022-02-01:
I mean STEP and CSG formats support cylinders and spheres, which makes arc generation possible.
CSG is simplified OpenSCAD export format.

supermerill, 2022-02-01:
And still, you can't slice & offset them.

fabseeeee, 2022-10-15:
**something new about this? .... i think is a great idea !!!!!**