supermerill/SuperSlicer #325: [Feature Request] Search Feature

MarcPot, 2020-06-29:
_Is this a new feature request?_ Yes

With all the awesome features and options you're adding the UI is getting a little crowded, so maybe a new feature could be a search and/or visibility setting?

Kinda like Cura, but the way Cura did it is really slow, so not like Cura. I think the problem with Cura is that it works like a filter on your current UI, making the whole UI reload every time (I'm not a coder so I could be completely wrong).

supermerill, 2020-06-29:
will be added when merged against prusa 2.3

thelandinlook, 2021-01-11:
Hello!

thelandinlook, 2021-01-11:
I've been looking for this, has it been added yet?

thanks xo

MarcPot, 2021-01-14:
Not sure if it's in the stable release yet since I don't use that one. But 
[image]

closed 2021-01-14 by MarcPot