supermerill/SuperSlicer #326: Flow calibration objects are too close eache others

beber75, 2020-06-29:
### Version
2.2.52

### Operating system type + version
Windows 10, build 19041.329

### 3D printer brand / version + firmware version (if known)
Custom 300x300x400 using klipper 0.8

### Behavior
I have a bltouch on the left side of the nozzle with this position in printer.cfg:

```
[bltouch]
x_offset: -50
y_offset: -10
```

When printing +20 and +10 calibration object, my bltouch is hitting the printed part on the left. You should definitively give more space between test objects to avoid these kind of problem.

_Is this a new feature request?_

No
[image]

beber75, 2020-06-29:
I figured I could adjust it before slicing but too late for me.

supermerill, 2020-06-29:
You have to modify your 'extruder clearance'. It's in the output tab.
I may create a more refined one int he printer setting , but it's not for now.

beber75, 2020-06-29:
My bad, thanks.

closed 2020-06-29 by beber75