supermerill/SuperSlicer #327: Small details lost on surfaces

dkatzdev, 2020-06-29:
### Version
2.2.52.0

### Operating system type + version
Windows 10 v1909

### 3D printer brand / version + firmware version (if known)
Creality Ender 5 Pro

### Behavior
When using the latest version of SS (2.2.52.0), small details are lost in the surface of objects. These details were maintained with identical settings in version 2.2.51.2. Likely related to the new thin wall changes in the slicer. See below

The model:
[image]

Sliced in 2.2.51.2
[image]

Sliced in 2.2.52.0
[image]

#### Project File (.3MF) where problem occurs

[filament_sensor_cover.zip](https://github.com/supermerill/SuperSlicer/files/4848122/filament_sensor_cover.zip)

supermerill, 2020-06-29:
try with 0 on "xy compensation - hole", it's a known issue on this release i haven't the time to fix yet. (see release page)

dkatzdev, 2020-06-30:
Yep, that fixes it, but I'd like to be able to maintain my hole adjustment for more complex parts.

dkatzdev, 2020-07-06:
Resolved in latest bugfix 2.2.52.1

closed 2020-07-06 by dkatzdev