supermerill/SuperSlicer #328: Thin Wall Problem

JoduWei, 2020-06-30:
### Version
Current revision + 2.2.52.0_win64_20200623

### Operating system type + version
Windows 10

### Behavior
See Screenshots

[image]

[a_drive_frame_lower.3mf.zip](https://github.com/supermerill/SuperSlicer/files/4848723/a_drive_frame_lower.3mf.zip)

supermerill, 2020-06-30:
https://github.com/supermerill/SuperSlicer/issues/150

n8bot, 2020-06-30:
The slicer is avoiding the overlap of the two external perimeters from the hole and the outside of the part, where they become so close as to require this. As @supermerill mentions in the issue linked, you need to enable the option to allow overlap of perimeters.

On the topic of the overlap of perimeters, Slic3r default behaviour is indeed to overlap the perimeters slightly.

From [Slic34 Manual —­ Flow Math](https://manual.slic3r.org/advanced/flow-math):
>Path spacing is thus:
>
>```spacing = extrusion_width - layer_height * (1 - PI/4)```

Is this behaviour still present in SuperSlicer? I'm not actually familiar with coding enough to understand everything and determine this answer myself. Sorry if this is a stupid question.

supermerill, 2020-06-30:
yes, but ince 2.2.52, you can change that for periemters with the perimeter overlap.

n8bot, 2020-06-30:
Oh, nice!

I guess this is what is meant by this note on the release page:

> - #4 perimeter overlaps : total control over perimeter width vs spacing. These settings may allow you to retain the same perimeter quality when you under-extrude but without under-extruding. Also may help to hide infill pattern passing through the perimeter. These needs some experimentation to see if they are useful.
>   - external_perimeter_overlap to control half of the overlap between the external perimeter and the inner one: 0% is no overlap, 100% is full overlap (current & default value)
>   - add perimeter_overlap, same between every perimeter (and the other half with the external one)
>   - add perimeter_bonding, that shrink the external-internal spacing, without growing the infill polygons.

Would you be able to point me to the area in the source code that this is done? Just the general direction if it's complex. I'm just curious, I don't have any intention on modifying anything.

supermerill, 2020-06-30:
periemtergenerator.cpp (in the begining) for the use of it.
flow.cpp for the multiplication.

closed 2020-08-25 by supermerill