supermerill/SuperSlicer #329: Support marking feature

Tumblebeer, 2020-06-30:
The method shown in this youtube video: https://www.youtube.com/watch?v=oBGl4i668Tg
Should be possible to include in this slicer, perhaps by allowing the "ironing" to be performed with a second extruder (which is a marker)? 
Or even by post-processing the gcode, if the ironing moves could be commented as "ironing" i verbose mode.

supermerill, 2020-06-30:
Right way to do that will be to add a new "extruder type" and then create a special "post-processing" option for it, a bit like the "milling" one.

yschroeder, 2020-07-09:
This looks interesting.

By the way there are click action sharpies available. Mount such a sharpie to the carriage and extend the tip by lifting z until it hits something to press the button of the pen.

This would require some gcode to be executed once the extruder should be changed.