supermerill/SuperSlicer #330: Extended Flow Control

afkfurby, 2020-07-01:
It would be nice to have the ability to control the flow of every entry where you can set the extrusion width and speed.
This way I can print inner perimeters at higher speeds while maintaining the correct flow.

supermerill, 2020-07-01:
Can you write what you have in your head in much more details?
why?
for what?
how?
Also try to replace some subjective words like "correct" by objective ones.

yschroeder, 2020-07-05:
My best guess on this: At high printing speeds, most setups tend to underextrude. He wants to increase the flow of certain extrusion types to mitigate the underextrusion.

However, the better way would be to have an extrusion multiplier that is dependent on the volumetric flow. This needs to be calibrated (might be something for the calibration section).

supermerill, 2020-07-05:
> At high printing speeds, most setups tend to underextrude

I don't understand, where does the plastic go? If we ask to extrude for 30mm, even if the speed increase, it will always be 30mm, (unless the extruder stepper fail, but it's not the slicer fault).

I may do something for retraction to have a better "pressure model", but here, i really does not understand the concerns / issues.

n8bot, 2020-07-05:
> > At high printing speeds, most setups tend to underextrude
> 
> I don't understand, where does the plastic go? [..]

I suppose that the extruder drive could be losing microstep positions. It really shouldn't add up to much motion lost, but I haven't analyzed it very deeply. The drive gear teeth could also slip over the filament if commanded too fast, and the idler tension is not very high.

This situation seems to be better handled by the printer firmware. Pressure Advance (or Linear Advance) would be best. RepRapFirmware also supports non-linear extrusion exactly like this: faster speed = more extrusion).

See: [RepRapFirmware nonlinear extrusion (M592)](https://duet3d.dozuki.com/Wiki/Gcode#Section_M592_Configure_nonlinear_extrusion)

yschroeder, 2020-07-05:
This video sheds some light on the topic:
https://www.youtube.com/watch?v=0xRtypDjNvI

Basically, when reach the extrusion limit of your setup, it tends to underextrude even if the extruder stepper is not losing steps. When it starts losing steps you will obviously underextrude, but even before the extruder gears start to slip on the filament.

The issue popped up in Klipper a few times already, but nothing happended yet:
https://github.com/KevinOConnor/klipper/issues/2610

I do not strictly agree that this is better handled by firmware, as extruder gears slipping over filament might differ with filament hardness. It might be something for the filament section of SuperSlicer.

That M592 seems like the solution to the problem, however, it is not widely supported by firmwares...

n8bot, 2020-07-05:
> [...]
> 
> I do not strictly agree that this is better handled by firmware, as extruder gears slipping over filament might differ with filament hardness. It might be something for the filament section of SuperSlicer.
> 
> [...]

RepRapFirmware has a filament system that allows the user to define settings on a per-roll basis, load and unload the selected filament and bring in settings.

IMO, slicer software should define the optimal toolpath, and firmware should correct it where neccesary. In some cases, this may not be possible, partly in part due to the limitations of GCode (I guess).

Regardless, you're right, there are other firmwares to support than RRF, so if it's deemed necessary, it should be included in the slicer.

afkfurby, 2020-07-07:
sorry, github stopped sending me notifications, yshroeder made the point, that's why I need this.
We have printers with nozzles bigger 1mm and speeds up to 100mm/s.
I noticed if an object with 2 perimeters is printed and 40mm/s outer and 80mm/s inner speed, the walls are not the same thickness, the inner wall is even underextruded. Because at those speed the melted material gets compressed in the melt zone and maybe also small amounts flow back up and fill all the small gaps. 
That's why :)

supermerill, 2020-07-07:
Did you set the maximum flow rate your extruder can output in your filament setting (15 for a volcano in pla)?

What  I understand everyone here ask for a compensation system to push their system well over their designed limits. I'm not sure if it's really something that should be in a slicer, and if it's desirable.

yschroeder, 2020-07-07:
It is not really pushing over the limits at this point. The underextrusion also exists at very low print speeds. See, the linked video. It just becomes much more problematic at higher speeds, thus pressures. If we would push the filament until the extruder loses steps, I would be with you that we are over the machine limits. But this happens before that point already.

n8bot, 2020-07-07:
There are other ways for the extruder to lose motion than skipping steps. Insufficient idler tension is common. I had to, myself, change from a spring-tensioned idler to a fixed-distance idler to remove all traces of filament slipping from my extruder. I do not get loss of motion, even at the threshold of the limit of extrusion rate. 

Like I said in the klipper thread, if your tires are spinning, how will stomping on the gas help? Sure, there will be a minuscule increase in velocity due to the tires spinning and creating friction, but it will be negligible and will never account for the lost motion -- the motion was lost for a reason! The best that could be hoped for would be to slightly reduce the underextrusion. You'd likely be extruding 200% the amount required, and still underextruding. Sensible? I think not.

But also, in terms of adding this to a slicer, it makes no sense. The actual speed that the toolpath is printed at is determined by the firmware. It can be limited for a variety of reasons. If you artificially increase the extrusion amount in slicer, but the firmware is also trying to limit the toolpath, you'll have VERY weird behaviour.

Also, how do you account for acceleration? If the extrusion amounts are adjusted in slicer, there is no way for the firmware to know that and the acceleration and deceleration phases will be overextruded.

yschroeder, 2020-07-07:
Yes after reading up on the topic, also on the marlin forum I agree that this slip is pressure dependent and thus the slicer cannot compensate it.

However, if part of the slip is due to the filament deforming when grabbed by the teeth of the extruder, this could be compensated by extra extrusion length. Still, the existence of this problem needs to be confirmed.

From my point this issue can be closed and discussion should continue in the linked Klipper issue.

dewi-ny-je, 2020-08-13:
> Like I said in the klipper thread, if your tires are spinning, how will stomping on the gas help? Sure, there will be a minuscule increase in velocity due to the tires spinning and creating [...]

You are approaching the issue the wrong way.
I assume you watched CNC's video. Have you see that the actual flow increases together with the slippage? If you are at 5% slippage, increasing the requested flow will bring you to the desired value.

Slippage is not on/off, and if you look at the graph starting from the Y axis, crossing the extrusion curve, and then going down to the X axis, there is quite a range of values where the desired extrusion can be achieved by increasing slippage.

So yes in theory you can increase the requested flow to account for slippage.

As we discussed, many more factors are in play, so I'm not saying that this feature would solve anything, but introducing it in SuperSlicer may allow to perform testing which could provide valuable information.

n8bot, 2020-08-13:
This literally can not be added to a slicer, it must be a firmware feature. This feature already exists in RRF, and users have reported that when trying it, they get either underextrusion or overextrusion -- I have not found a user reporting that they have achieved a "correction" of whatever behaviour they were trying to correct.

You may be able to account for some of the lost movement by just slamming the filament through faster, but it's doubtful. test it yourself with RRF.

dewi-ny-je, 2020-08-15:
Sure, a proper implementation should be done on firmware. My remark was only about the concept "if you are having slippage you cannot get the desired filament flow by pushing more" which, just by looking at the published graph, is wrong (within a certain range). If you have 3% underextrusion you can push more to compensate. Of course if you have 10% underextrusion you are outside the acceptable range and your remark becomes correct.

That said, the original proposal was to add a extrusion flow compensation per each line width.
I see little damage in that. In long extrusion lines, which infill typically uses, it would perform the same thing that a firmware correction would. It would be an easy way to test the idea.

n8bot, 2020-08-15:
I don’t really put much faith into that published graph. It’s one small data point, and it shows more than anything that the concept won’t work. 

Also, this can not be put into a slicer because acceleration and starting speed need to be accounted for. 

In fact, this is where I think the failure actually occurs: acceleration. So, by increasing the top speed, you’ll increase the top acceleration, and I think the filament will still underextrude if you saw underextrusoion at lower speeds. 

Think about the physics of this. Acceleration requires much more force than maintaining a coasting speed. 

I don’t personally see the effects that people want this feature to account for, so I can’t test it myself. 

In the end, it literally can not be added to a slicer without causing sever overextrusion at the beginning and end of affected paths.

dewi-ny-je, 2020-08-15:
I agree on the likely overextrusion at the beginning and end when the slicer tries to change the extrusion factor, but not about the fact that the issue of underextrusion is caused by the acceleration: you can see from this microphotograph http://www.extrudable.me/2013/04/18/exploring-extrusion-variability-and-limits/ that the marks on the filament get closer to each other when pushed at high speed, showing therefore that the issue is not related to a loss of steps but to a plastic deformation of the filament as result of the high force applied by the gears to the filament surface (a very small contact surface).
Doubling the gears (BMG) increases the contact surface and reduces the issue, but it doesn't solve it. Having a sort of a continuous track (the one used in tanks) would basically eliminate the issue, but no extruder I'm aware of offers it.

dewi-ny-je, 2020-08-15:
That said, having a separate extrusion multiplier for infill would be useful also to strengthen it, independently from the high speed underextrusion issue