supermerill/SuperSlicer #332: Custom Support Enforcers or Blockers are ignored

afkfurby, 2020-07-02:
### Version
2.2.52

### Operating system type + version
Windows 10 Build 1909

### 3D printer brand / version + firmware version (if known)
irrelevant

### Behavior

* _Describe the problem_
Custom Support Enforcers/Blockers loaded from an STL are ignored
* _Steps needed to reproduce the problem_
load custom support from an stl
* _Expected Results_
support placed or not placed
* _Actual Results_
 nothing, completly ignored

afkfurby, 2020-07-02:
[support error.zip](https://github.com/supermerill/SuperSlicer/files/4863887/support.error.zip)

supermerill, 2020-07-02:
move them like ~0.2mm up in z.

Support enforcer/blocker work only on the 'interface' volume, the rest is useless.

afkfurby, 2020-07-03:
ahh thanks, a hint for that would be good :) otherwise, really good work. thanks.

closed 2020-07-03 by afkfurby