supermerill/SuperSlicer #333: Precompiled MacOS SuperSlicer attempting to run FreeCAD.app/python.exe

sleeperninja, 2020-07-03:
### Version
SuperSlicer_2.2.52

### Operating system type + version
MacOS Catalina 10.15.5 (19F101)

### 3D printer brand / version + firmware version (if known)
MMSP running ADVi3++ 4.0.6

### Behavior
#### SuperSlicer FreeCAD Python Script attempts to run python.exe
_In order to reproduce, click Generate, FreeCAD Python Script, then Generate._
**Expected results**: output of default script
**Actual results**: "Error, cannot find the freecad (version 0.19 or higher) python at './Applications/FreeCAD.app/python.exe', please update your freecad bin directory in the preferences."
[image]

#### This is not a new feature request.

#### Project File (.3MF) where problem occurs
This error is project independent.

n8bot, 2020-07-03:
Do you have FreeCAD 0.19 or higher installed?

From the [release page:](https://github.com/supermerill/SuperSlicer/releases/tag/2.2.52.0)
>[...] need to have FreeCAD 0.19 installed.

supermerill, 2020-07-03:
ohh, right, i forgot about other os.
I'll fix thgat

sleeperninja, 2020-07-06:
If I add the path to FreeCAD.app, it automatically searches for /bin/python, rather than .../Contents/Resources/bin/python

`Error, cannot find the freecad (version 0.19 or higher) python at '/Applications/FreeCAD.app/bin/python', please update your freecad python path in the preferences.
`
If I specify /Applications/FreeCAD.app/Contents/Resources/bin/python, I can tell it likes the path _better_, but it still fails to properly generate.

```
Python 3.8.3 | packaged by conda-forge | (default, Jun  1 2020, 17:21:09) 
[Clang 9.0.1 ] on darwin
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'FreeCAD'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'Part'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'Draft'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/sleeperninja/Library/Application Support/SuperSlicer/scripts/FreePySCAD/freepyscad.py", line 27, in <module>
    import FreeCAD, Part, math, Draft, InvoluteGearFeature, importSVG, Mesh
ModuleNotFoundError: No module named 'FreeCAD'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'App' is not defined
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'set_font_dir' is not defined
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 1, in <module>
NameError: name 'scene' is not defined
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'Mesh' is not defined
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'App' is not defined

Error, no object generated.
```

supermerill, 2020-07-07:
Erf, if the freecad python has different behavior for each os...
I'll take a look in my ubuntu vm.

sleeperninja, 2020-07-10:
[image]
It doesn't produce an error, but do note that I used *sys.path.append('/Applications/FreeCAD.app/Contents/Resources/lib')* as opposed to *sys.path.append('/Applications/FreeCAD.app/Contents/lib')*

closed 2020-10-22 by supermerill