supermerill/SuperSlicer #334: xy compensation removes hole completely

AmyTheCute, 2020-07-05:
### Version
2.2.52

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
The Doot changer, Klipper

### Behavior
XY Compensation makes a hole completely go away in one of my VERY similar parts but not in the other.
[image]

[image] 
I've included two photos. One with 0.0mm compensation and another with 0.05mm compensation. happens with any value. the right part is just fine while the left one gets the inner/smaller hole completely removed

#### Project File (.3MF) where problem occurs

[GitHub ISssue.zip](https://github.com/supermerill/SuperSlicer/files/4874811/GitHub.ISssue.zip)

supermerill, 2020-07-05:
duplicate of #327 and #317 
fix in 2.2.52.1

AmyTheCute, 2020-07-06:
Oh I'm sorry! I did google but I didn't find those

closed 2020-07-06 by AmyTheCute