supermerill/SuperSlicer #335: slicer ignores features

Coffee0297, 2020-07-07:
### Version
2.2.52
### Operating system type + version
win10
### Behavior
when slicing this the TS100_Top_ file don't get the internal parts with, prusa slicer do with same profile

[image]
[image]

#### Project File (.3MF) where problem occurs

[OctoPCB-Alligator-TS100.zip](https://github.com/supermerill/SuperSlicer/files/4887179/OctoPCB-Alligator-TS100.zip)

supermerill, 2020-07-07:
seems like duplicate of duplicate of #334, #327 and #317
fix is in 2.2.52.1

unless it's in 2.2.52.1

neophyl, 2020-07-08:
Just checked the attached 3mf in 2.2.52.1 and the slice looks ok to me.  All the parts are there.

closed 2020-08-24 by supermerill