supermerill/SuperSlicer #336: Feature Request: Line Preview

quojure, 2020-07-09:
Version
2.2.52

Win10 64

Feature Request:

Thanks for the great fork.

Wondering if it would be possible to have a line by line preview similar to Cura/S3D on the preview screen. That way you can see the order of every line segment in a layer, whereas now you can only view layer by layer. Good for fine tuning retraction settings and troubleshooting. I am running the gcode through S3D to check the animation as a work around.

neophyl, 2020-07-09:
That feature is currently being added to the Prusa 2.3.0 slicer.  Given that fact and with Super Slicer being built on that as a base then in due course I would expect it to also appear in a future version on Super Slicer too.
So just give it time :)

yschroeder, 2020-07-09:
Would this also include the visualisation of gcode not generated by the slicer? This would be very handy to debug/investigate other gcode.

supermerill, 2020-07-09:
From the prusa livestream, yes.

DeviousPenguin, 2020-07-10:
@yschroeder you can do this in Cura already, you need to go to 'preview' then open your gcode.

It would be really nice to have this in PS/SS also however, as Cura tends take forever to load on my machine and I much prefer the UI of Slic3r and it's forks :)

quojure, 2020-07-10:
Thanks - just saw the livestream. I have just had a play around with the 2.3.0-alpha0 release. Looks like it is not yet in there, so hopefully will be included in the next alpha release.