supermerill/SuperSlicer #337: GLIBC not found

Goury, 2020-07-11:
### Version
2.2.52.1

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Irrelevant

### Behavior
Execute SuperSlicer-ubuntu-2.2.52.1.AppImage, got this:

/tmp/.mount_SuperS2VhQnS/bin/superslicer: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_SuperS2VhQnS/bin/superslicer)
/tmp/.mount_SuperS2VhQnS/bin/superslicer: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_SuperS2VhQnS/bin/superslicer)
/tmp/.mount_SuperS2VhQnS/bin/superslicer: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.mount_SuperS2VhQnS/bin/superslicer)

#### Project File (.3MF) where problem occurs
Irrelevant

Come on, the point of having an appimage is to not require any external requirements.

supermerill, 2020-07-11:
I compiled it on 20.04. Can you try with https://github.com/supermerill/SuperSlicer/suites/902405721/artifacts/10836927, as this one should be compiled on 18.04?

> Come on, the point of having an appimage is to not require any external requirements.

....

Goury, 2020-07-11:
> bit more understanding
No. I am understanding enough.
This is why I gave it some effort and then some more to file this issue.

How's about instead you learn how to package appimages properly (or better yet, a deb package for every supported distro out there, it isn't that hard, really) and I'll tell every single one of my 3d printing friends that Prusa truly sucks ass and your fork is superior?

Because it does have a lot of features but, sadly, QA department could do better job.

> Can you try

It seems to work fine, except for these warnings:

(superslicer:10267): GLib-GObject-WARNING **: 04:49:22.340: g_object_weak_unref: couldn't find weak ref 0x7f8f63770de0(0x5582f9f95090)

(superslicer:10267): GLib-GObject-WARNING **: 04:49:22.421: g_object_weak_unref: couldn't find weak ref 0x7f8f63770de0(0x5582f9051310)

(superslicer:10267): GLib-GObject-WARNING **: 04:49:22.481: g_object_weak_unref: couldn't find weak ref 0x7f8f63770de0(0x5582f82a24b0)

closed 2020-07-11 by supermerill