supermerill/SuperSlicer #338: Compilation failed: error: no type named ‘executor_type’ in ‘class boost::process::detail::posix::async_pipe’

stelgenhof, 2020-07-11:
### Version
2.2.52.1-5-gbe17f49cd

### Operating system type + version
Arch Linux  5.7.7-arch1-1

### Behavior
Compilation did not succeed due to the following error:
```
/code/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:780:114:   required from here
/usr/include/boost/asio/impl/read.hpp:719:53: error: no type named ‘executor_type’ in ‘class boost::process::detail::posix::async_pipe’
  719 |     typedef typename AsyncReadStream::executor_type executor_type;
      |                                                     ^~~~~~~~~~~~~
[ 78%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Gizmos/GLGizmoRotate.cpp.o
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:334: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/FreeCADDialog.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1471: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:161: all] Error 2
```

This seems to be related to the Boost library (after having a look around the web). Boost library I am using is 1.72

supermerill, 2020-07-11:
deps should compile boost 1.70

witem, 2020-08-14:
OS:  5.7.9-1-MANJARO

I have the same problem. cmake output:
```
/path/SuperSlicer/build (master ✔) ᐅ cmake  -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr -DSLIC3R_FHS=1 -DSLIC3R_GTK=3 -DSLIC3R_BUILD_TESTS=OFF  ..
-- The C compiler identification is GNU 10.1.0
-- The CXX compiler identification is GNU 10.1.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_FHS: 1
-- SLIC3R_GTK: 3
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.64.0") found components: system filesystem thread log locale regex chrono atomic date_time
-- Boost::boost exists
-- Found TBB: /usr/include (found version "2020.2")
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /usr/include
--   TBB_DEFINITIONS         =
--   TBB_LIBRARIES           = /usr/lib/libtbb.so
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     =
--   TBB_DEFINITIONS_RELEASE =
--   TBB_LIBRARIES_RELEASE   = /usr/lib/libtbb.so
-- Found CURL: /usr/lib/libcurl.so (found version "7.71.1")
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.11")
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.3")
-- Found EXPAT: /usr/lib/libexpat.so (found version "2.2.9")
CMake Warning (dev) at /usr/share/cmake-3.17/Modules/FindOpenGL.cmake:305 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMakeLists.txt:391 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/libOpenGL.so
-- Found GLEW: /usr/include (found version "2.2.0")
-- Found NLopt in '/usr/lib'.
-- Using NLopt include directory '/usr/include'.
-- Using NLopt library '/usr/lib/libnlopt.so'.
-- Found OpenVDB: /usr/include (found suitable version "7.0.0", minimum required is "5.0") found components: openvdb
-- OpenVDB ABI Version: 7
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0") found components: iostreams system
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Performing Test LLVM_HAS_ATOMICS
-- Performing Test LLVM_HAS_ATOMICS - Success
-- OpenVDB libraries: /usr/lib/libopenvdb.so
-- Miniz NOT found in system, using bundled version...
-- Using qhull from system.
-- [libigl] Looking for Eigen3
-- IGL found, using system version...
-- __cplusplus is 201402
--   --> Do not link with Boost.Thread
-- Found Boost: /usr/lib64/cmake/Boost-1.72.0/BoostConfig.cmake (found suitable version "1.72.0", minimum required is "1.48")
-- Boost include dirs: /usr/include
-- Boost libraries:
-- wx-config path: /usr/bin/wx-config
-- wx libs: -pthread;;;-lwx_baseu-3.1;-lwx_gtk3u_core-3.1;-lwx_gtk3u_html-3.1;-lwx_gtk3u_gl-3.1;-lwx_gtk3u_stc-3.1
-- Configuring done
-- Generating done
-- Build files have been written to: /path/SuperSlicer/build
```

stelgenhof, 2020-08-14:
Arch Linux (and distros like Manjaro) have a rolling release meaning always the latest stable version of packages are installed. I think there are a couple of options:
- Downgrade Boost package on Arch Linux to 1.70
- SuperSlicer upgrades Boost to 1.72

Would also be good if there was some kind of warning upfront if the requirements are not met.

Cheers! Sacha

supermerill, 2020-08-16:
There should be a third way:

 * compile boost 1.70 and link it statically.

witem, 2020-08-16:
> There should be a third way:
> 
>  * compile boost 1.70 and link it statically.

Can you share some instructions how to do this?

supermerill, 2020-08-16:
my script for building on linux: https://github.com/supermerill/SuperSlicer/blob/master/.github/workflows/ccpp_ubuntu.yml

basically, execute each command after "run"

witem, 2020-08-16:
> my script for building on linux: https://github.com/supermerill/SuperSlicer/blob/master/.github/workflows/ccpp_ubuntu.yml
> 
> basically, execute each command after "run"

thanks, but this script not compile differnt version of `boost` and don't link it statically =)

n8bot, 2020-08-16:
I think the following command would be what "links it statically," but I'm also just a newb.

`cmake .. -DCMAKE_PREFIX_PATH="/home/runner/work/SuperSlicer/SuperSlicer/deps/build/destdir/usr/local" -DSLIC3R_STATIC=1`

Specifically, the `DSLIC3R_STATIC=1`

supermerill, 2020-08-16:
also, if you look at the https://github.com/supermerill/SuperSlicer/blob/master/deps/deps-linux.cmake that is used by the "cmake .." from deps/build directory, it should compile a 1.70 version of boost.

stelgenhof, 2020-08-17:
I tested it once more but still it used version 1.72. Checking the output, it doesn't seem the `deps-linux.cmake` is being used/loaded. I couldn't find anything related to 1.70 in the outputs.

I can share my outputs/logs if that may be helpful.

supermerill, 2020-08-17:
Yes, with the complete history of the commands executed.

witem, 2020-08-17:
@supermerill @stelgenhof Thank you for your help.
 
For build deps and links it statically wroks. Maybe need add it to docs. 
My steps:

1. cd to root SuperSlicer
1. `mkdir deps/build` and `mkdir build`
1. `cd deps/build`
1. `cmake ..`
1. `make`
1. `cd ../../build`
1. `cmake  -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr  -DCMAKE_PREFIX_PATH="/home/user/projects/SuperSlicer/deps/build/destdir/usr/local" -DSLIC3R_STATIC=1 -DSLIC3R_WX_STABLE=1 -DSLIC3R_STATIC=1 ..`
1. `make` - in this step I have error on 96% in tests (`/SuperSlicer/tests/superslicerlibslic3r/test_print.cpp:10:10: критична помилка: libslic3r/config.hpp: Немає такого файла або каталогу`) but when I commented tests in `build/CMakeFiles/Makefile2` make finished success
1. `sudo make install`

stelgenhof, 2020-08-17:
@witem Thanks for the steps. Let me give that a try.

stelgenhof, 2020-08-17:
@supermerill @witem I was able to compile it successfully with the same steps as @witem. Also same here had the tests failed at 97%: 
```
SuperSlicer/tests/superslicerlibslic3r/test_print.cpp:10:10: fatal error: libslic3r/config.hpp: No such file or directory
   10 | #include <libslic3r/config.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
When commenting these out as @witem mentioned, compilation was successful.

Cheers! Sacha

closed 2021-06-02 by supermerill