supermerill/SuperSlicer #339: Conditional gcodes

appjaws, 2020-07-11:
### Version
2.2.52

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Home made Core XYUV running Duet firmware

### Behavior
This is my start gcode I use with Simplify3D
M98 P"0:/macros/extra/Set LED Red"
if !move.axes[0].homed || !move.axes[1].homed
     G28
G28 Z
;M98 P"0:/macros/Conditional/Auto_Bed_Levelling"
M98 P"0:/macros/Auto Nozzle Cleaning"
M98 P"0:/macros/extra/Set LED Blue"
G1 Z5
T0

The slicer stops with an error at the if statement. This is the error:-
!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 2. Expecting a legacy variable expansion format
if !move.axes[0].homed || !move.axes[1].homed
              ^
!!!!! End of an error report for the custom G-code template start_gcode

Is it possible for the SuperSlicer to just add this code without an error?

Thanks for your great work, really getting into SuperSlicer now.

supermerill, 2020-07-11:
you have to use this macro syntax
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

appjaws, 2020-07-11:
Thank you for the quick reply.

https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

closed 2021-04-27 by supermerill

CCS86, 2021-05-18:
What was the solution here?

The link provided is a blank page now.

supermerill, 2021-05-20:
it's now here https://help.prusa3d.com/en/article/macros_1775

CCS86, 2021-05-20:
> it's now here https://help.prusa3d.com/en/article/macros_1775

Thanks for the link. I had found that page, but still haven't found a workaround for the original issue.

Most simply RRF allows conditional gcode, so I just want to pass these lines of code in my SS start gcode block. But SS assumes I want this code parsed and run using its own conditional code/macro support.

Is there some sort of an escape character that forces SS to simply pass the code as written?

supermerill, 2021-05-20:
`if !move.axes[0].homed || !move.axes[1].homed`
=>
`{"if !move.axes[0].homed || !move.axes[1].homed"}`

CCS86, 2021-05-21:
Oh man, I was so close! I had the curly brackets and quotes, but I had square brackets wrapped around all that.

Thank you much!