supermerill/SuperSlicer #340: Object centers of cloned objects are the same as parent (Label Output feature)

paukstelis, 2020-07-11:
### Version
2.252

### Operating system type + version
Linux
### 3D printer brand / version + firmware version (if known)
Any

### Behavior
I was preparing to (finally) include object center detection for Octoprint CancelObject plugin. I noticed that the object centers for all sub-instances of an object return the same object and bounding box origin. Separate object instances work as expected.

paukstelis, 2020-07-11:
Actually, as I look closer, it seems that the object centers and bounding boxes are derived from the positions of all of the cloned models.

paukstelis, 2020-08-06:
Testing this out and noticed that Y value of bounding box center is the same for all 3 instances of an object (when they shouldn't be).
```
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 0","object_center":[110.500000,90.500000,110.500000],"boundingbox_center":[90.500000,10.000000,20.000000],"boundingbox_size":[20.000000,20.000000,0.000000]}
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 1","object_center":[110.500000,119.500000,110.500000],"boundingbox_center":[119.500000,10.000000,20.000000],"boundingbox_size":[20.000000,20.000000,0.000000]}
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 2","object_center":[139.500000,105.000000,139.500000],"boundingbox_center":[105.000000,10.000000,20.000000],"boundingbox_size":[20.000000,20.000000,0.000000]}
```

supermerill, 2020-08-06:
weird.
object_center and boundingbox_center should be the same, with the z at 0 for the object one.

supermerill, 2020-08-27:
@paukstelis can't reproduce your resutl, can you send me your .3mf?

paukstelis, 2020-08-27:
[20mm_cube.zip](https://github.com/supermerill/SuperSlicer/files/5136737/20mm_cube.zip)
3MF as zip.

On 2.2.53 I still get this:
```
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 0","object_center":[110.500000,90.500000,110.500000],"boundingbox_center":[90.500000,10.000000,20.000000],"boundingbox_size":[20.000000,20.000000,0.000000]}
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 1","object_center":[110.500000,119.500000,110.500000],"boundingbox_center":[119.500000,10.000000,20.000000],"boundingbox_size":[20.000000,20.000000,0.000000]}
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 2","object_center":[139.500000,105.000000,139.500000],"boundingbox_center":[105.000000,10.000000,20.000000],"boundingbox_size":[20.000000,20.000000,0.000000]}
```

supermerill, 2020-08-27:
That's the output i get:
```
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 0","object_center":[110.500000,90.500000,0.000000],"boundingbox_center":[110.500000,90.500000,10.000000],"boundingbox_size":[20.000000,20.000000,20.000000]}
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 1","object_center":[110.500000,119.500000,0.000000],"boundingbox_center":[110.500000,119.500000,10.000000],"boundingbox_size":[20.000000,20.000000,20.000000]}
; object:{"name":"20mm_cube","id":"20mm_cube.stl id:0 copy 2","object_center":[139.500000,105.000000,0.000000],"boundingbox_center":[139.500000,105.000000,10.000000],"boundingbox_size":[20.000000,20.000000,20.000000]}
```

paukstelis, 2020-08-27:
Weird. I'm using Linux appimage build. I wonder if that is the issue?

supermerill, 2020-08-27:
I've put a new release, try with this one

paukstelis, 2020-08-27:
Same result. I've tried to build several recent versions of the master branch and it fails on some of the FreeCad stuff. (I'll open a separate issue for that)

supermerill, 2020-08-27:
just pushed the fix on master

paukstelis, 2020-08-27:
Closer. Now:
`GUI_ObjectList.cpp:2032:95: error: ‘spCustom’ was not declared in this scope; did you mean ‘erCustom’?`

paukstelis, 2020-08-27:
Compiles. Same issue with bounding box center. Maybe I can throw in some debugging to see why those values are different.

supermerill, 2020-08-27:
gcode.cpp l 1306
there should be 3 instance with 3 different array in printinstance.shift

paukstelis, 2020-08-27:
Bingo. Fixed it.
l 1313: 0 -> 0.0.
I guess this is required for %f.

supermerill, 2020-08-27:
oh, dang.
and as the implementation of the stdlib is different in every compiler...

closed 2020-10-22 by supermerill