supermerill/SuperSlicer #341: Switching from PrusaSlicer

pedjas, 2020-07-13:
### Version
2.2.52

### Operating system type + version
Windows 8.1 

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S

### Behavior
I just installed SuperSlicer. I use PrusaSlicer.  Is there option to get all PrusaSLicer configuration copied (and maybe synced with PrusaSlicer). I manged to copy only my custom profiles, but default profiles from Prusa (printers and filaments) are not there.

My goal is to have SuperSlicer prepared in the same manner as PrusaSlicer. Right now I only have some BIBO2 printer ant that is all.

_Is this a new feature request?_
Do not know.

supermerill, 2020-07-13:
you can rename you config directory fromprusaslicer to SuperSlicer.
It"s locaed at c:/users/myname.appdata/roaming

pedjas, 2020-07-14:
I copied everything from PrusaSlicer to SuperSlicer but PrusaSLice.ini. When I copied that and renamed to SuperSlicer.ini, I got everything in order.

I do not get synced PrusaSlicer and SuperSlicer. I will deal with that later, after I get used to SuperSlicer.

closed 2021-04-27 by supermerill