supermerill/SuperSlicer #342: Variable thread width

espr14, 2020-07-13:
0.4 mm E3D V6 nozzle can print up to 1 mm width thread since it has flat cut. Wider thread means higher strength and higher speed while narrow thread provides higher precision. Current implementation allows fixed width which is not flexible.

It would be nice to have a range of widths (min, max) and a threshold parameter. This parameter would represent tendency to use more or less narrower thread according to angle of a corner. On straight lines it would use maximal width. Transition between wider and narrower thread has to be smooth because it changes the path of a nozzle.

yschroeder, 2020-07-15:
This would be REALLY nice! One thick perimeter on long smooth sections (they don't have to be straight) and 2 or 3 thin perimeters at corners to get a sharp corner. Also, at overhangs it should do thin perimeters to avoid sagging.

supermerill, 2020-10-01:
You should add some schema / drawing and more explanation o how it works, as going from a wide to a narrow path currently create gaps.

n8bot, 2020-10-01:
[image]
[image]
More info: https://hal.inria.fr/hal-02568677v2/document
Source code: https://github.com/mfx-inria/Variable-width-contouring

The approach taken here talks about varying speed to achieve varying width, but that doesn't necessarily have to be the only way to use their methods.

supermerill, 2020-10-01:
I think the op wants the opposite.

n8bot, 2020-10-01:
Yeah, but the opposite doesn't really make sense. Increasing density of lines in the corners? Uhhh. I spent a lot of time playing railroad video games, and tight radii don't go well with multiple parallel tracks! :P

espr14, 2020-10-01:
1st example:

We have a border of a model with a tiny detail (top line). We want to use thicker thread (higher speed, higher strength). If we set thicker thread the detail is not printable in one go. So we can decrease the width for a while and then increase it like in this picture (top-down look). Arrow shows direction of printing.

[image]

2nd example:

We have a bridge (start is denoted by dashed line) so we have to use natural width (diameter of nozzle). Thus, we need to decrease width just before bridge overlap area (a little sooner than in previous example). There are 2 perimeters. The inner one can decrease width sooner (in the picture) or little later (maybe more complicated). There is a bridge infill for illustration.

[image]

yschroeder, 2020-10-02:
The second example from above would also fix the problem described in #407 

If I find some time later I will try to come up with drawings of some more examples.

In many cases this would also improve gap filling. By setting a `target wall thickness` and minimum/maximum extrusion width the hotend is capable of, the slicer could decide to use different width for the (external) perimeter based on the geometry.

Examples:
* `target wall thickness` would result in two perimeters with gap fill: increase perimeter width to avoid gap fill
* `target wall thickness` cannot fit two external perimeters in small feature: reduce perimeter width to allow fitting the two external perimeters (use the existing thin wall improvements of SuperSlicer if the feature is still too small)
* `target wall thickness` would result in round corners at sharp turns (e.g. 90°): reduce the external perimeter thickness to create a sharp corner

In cases where the `target wall thickness` cannot be achieved with one perimeter, add additional internal perimeters until `target wall thickness` is achieved.

yschroeder, 2020-10-02:
Here is one more example:

[image]

The perimeter width is configured to be 1mm. Normally, this would result in a corner radius of .5mm (left). However, if the nozzle diameter is smaller (here: 0.4mm), the perimeter width can be reduced when approaching the corner, resulting in a corner radius of 0.2mm.

kursatu, 2020-10-06:
I don't think this will work. This is because the extruder resolution will not be sufficient.
The idea is , if you have more curvature at the path, you decrease the line width.
I first thought that it might be super useful. 
But then I ran some numbers.
For a 2mm diameter feature (say a half circle), thin (0.3mm) vs. thick (0.5mm) line width,
at 0.1mm layer height, has a difference of only 3.5 microsteps. The extruder may not
really move actually. So you wont have the extrusion you need.
The spreadsheet with calculations is attached.
One can(?) devise a test, using a mix of straight, circular paths and manually change the
extrusion/line width at the circles. It will show the feasibility and usefulness.
It is hardly measurable though.
I have a 6200 steps/mm extruder and I can "print" 0.1 mm 1 line wide, straight walls from the printer tests.
But, you may need more features, even from firmware.
[Variable line width.zip](https://github.com/supermerill/SuperSlicer/files/5331654/Variable.line.width.zip)

espr14, 2020-10-07:
1. We are not printing separate objects so the extruder will be extruding continuously. Since the nozzle is kind of buffer, you don't need too high extruder precision.
2. Ratio between thickest and thinnest thread is up to 3 not just 0.3/0.5 (1.666).