supermerill/SuperSlicer #343: [BUG] Iron Pattern Generation

thierryzoller, 2020-07-13:
# Latest RC

1.  Gives me error the head will hit the pieces 
2.  Place them like on the picture (I believe that is not ideal placement) :)

Attached 3MF file.
[image]
[over_bridge_flow_test.zip](https://github.com/supermerill/SuperSlicer/files/4913939/over_bridge_flow_test.zip)

supermerill, 2020-07-13:
For the calibration, the "complete individual objects" is activated. So each object have to be at least at "extruder clearance" mm apart of each other.

thierryzoller, 2020-07-13:
I understand that - but why isn't it positioning them so that there is enough clearance?

supermerill, 2020-07-14:
It should position them so that there is enough clearance.
I used your project file, went into the ironing calibration, and press the generate, and there is no error.

supermerill, 2020-07-14:
[image]

thierryzoller, 2020-07-14:
I get the above result as soon as I press generate

neophyl, 2020-07-14:
I also get the same correct operation with your 3mf file.  Windows 10, Super Slicer v2.2.52.1

thierryzoller, 2020-07-14:
Well I guess not all the paramters superslicer thakes into account are within the file then. Let me make a video, is there a debug build?

n8bot, 2020-07-14:
Which OS are you using?

thierryzoller, 2020-07-14:
Windows 10 / 64bit

yschroeder, 2020-07-15:
I have also observed, that the placement of the generated objects is invalid. In my case they just did not fit on the print area (shown in blue). Clicking the auto arrange button at the top fixed it though. So maybe the calibration generator should trigger the auto placement automatically?

supermerill, 2020-07-15:
>  So maybe the calibration generator should trigger the auto placement automatically?

I'll try to trigger it if the bed is too small for the standard placement.

closed 2021-06-26 by supermerill