supermerill/SuperSlicer #345: Cutting corners generate invalid extrusion

kursatu, 2020-07-15:
### Version
_Version of SupserSlicer used goes here_
2.2.52
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
Cutting corners generate invalid extrusion lengts.
* _Steps needed to reproduce the problem_
At the Print Settings/Width & Flow tab, set the "Cutting corners" to %100.
Slice 
You;ll see this
G1 X162.679 Y198.665 E-nan(ind)

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

[String rainbow.zip](https://github.com/supermerill/SuperSlicer/files/4922151/String.rainbow.zip)

supermerill, 2020-08-01:
I've reworked it to use a stronger math formula.

kursatu, 2020-08-02:
Thanks. I've built with the latest code, and the nan in the gcode is not there anymore.
So the problem is fixed.
I will test it more as to how effective this feature is.
I have a very high resolution extruder, 4333 steps/mm, not that would position that precisely.
I also have 0.9degre x&y motors on my corecy printer.

supermerill, 2020-08-02:
> I have a very high resolution extrude

does the 5-digit precision of the gcode is enough for you?

n8bot, 2020-08-03:
I use ~4600 steps/mm on E, and I definitely benefited from increasing the precision of E values.

Speaking of the precision of gcode output, I just discovered today the solution to my "inconsistent extrusion rate problem." It was the XYZF_NUM precision. For some reason, rounding to 3 digits made VERY bad math. One value went from `Y-0.751` to `Y-0.758857000` with precision set to 9 instead of 3. That's quite far off.  No wonder the E value is not correct for the given vector norm.

I don't know how the rounding of that value gets so far off with the `std::fixed << std::setprecision`, but it does. Do you have any idea why that would be? Is it related to the fact that XYZF_NUM and E_NUM are both constantly changing the output stream precision? There is an instance in the gcodewriter.cpp file where XYZF_NUM is not used when outputting an F value. That can't be contributing to the error, can it?

Anyway, sorry to derail your topic again.

kursatu, 2020-08-10:
I am observing ragged edges with curves. Disabling the feature makes it go away.
Not sure what causes it, I have S-curve acceleration enabled on Marlin if it matters.
Attached is a hydrofoil that I am printing. It happens with other models as well.
[image]

supermerill, 2020-08-11:
I'll look at it.
It shouldn't even activate at these shallow angles.

closed 2020-10-22 by supermerill