supermerill/SuperSlicer #346: The speed does not change for some overhangs

kursatu, 2020-07-15:
### Version
_Version of SupserSlicer used goes here_
2.2.52
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Printer: Custom corexy printer, called Vking 
Firmware: MArlin bugfix-2.0.x
### Behavior
* _Describe the problem_
The speed for overhangs needs to be the same as the bridge speed, but I is not for my case.
It changes for some overhangs but not for all.
Please see the posted pictures
* _Steps needed to reproduce the problem_
Use the attached 3mf file and slice.
View the feature type at height 3.2mm
View the at its speed
[String rainbow.zip](https://github.com/supermerill/SuperSlicer/files/4927161/String.rainbow.zip)
[image]
[image]

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-07-15:
Is it problematic?
I have to test to see what's best, changing this speed can have some side effect on the perimeter "surface".

kursatu, 2020-07-16:
This feature is great. This particular case is probably not very problematic. 
The overhangs cause the filament curling up and causes print failure, at long prints with 20-100 pieces at once.
I have little bit of hard time getting  this feature to work as I expected.
For small layer heights, like 0.1 even though I've used 1% or 0.01mm, 
"As bridge threshold" width, it did not create overhangs.
These pictures below are at 0.3 and 0.1 layer height respectively, both with 0.01mm 
[image]
"As bridge threshold".

kursatu, 2020-08-10:
Here is another case that shows this is a problem.
This is  with very high thresholds that still does not trigger the bridge speed.
This is one of the most important features and would be good if it worked better.
[image]
[image]
[image]

kursatu, 2020-08-10:
This model is here and the bridge speed is 30mm/sec.
https://github.com/kickstarter/kickstarter-autodesk-3d

supermerill, 2020-11-01:
Should be fixed now,  just have to test it to be sure.

closed 2022-03-23 by supermerill