supermerill/SuperSlicer #347: Said to comment on perimeter overlap and why would need

robthide37, 2020-07-17:
https://github.com/KevinOConnor/klipper/issues/3025
if you look at last few posts we are having interesting find that the overlap is usually not ever a full step and just causes so called noise . The smarter members aka not me actually do reverse math with the flow formulas to figure out how to not have it messed with . They believe that turning off overlap will increase quality greatly on a perfectly tuned printer but also need turn off the extrusion modifier . I don't completely understand what is going on but i know you will.

supermerill, 2020-07-17:
I'll be honest, I don't have the will to read all this issue's posts. I don't understand why a slicer should bother with x/y "full step", it depends on the model, the printer, etc, it's impossible to target it anyway.

That said, if you want 0 overlap (and be prepared to have an abysmal bonding), you can put 0% to all two perimeter overlap. There will still be overlap in solid infill, but is it also problematic?

n8bot, 2020-07-17:
> That said, if you want 0 overlap (and be prepared to have an abysmal bonding), you can put 0% to all two perimeter overlap. There will still be overlap in solid infill, but is it also problematic?

I would like to have that 0 overlap in combination with non-adjusted path flow, so that the bonding will not be "abysmal."

In other slicers, I use 0% infill perimeter overlap and I get perfect bonding. No reason to overlap any paths ever.

robthide37, 2020-07-17:
the reason is that overlap is only nec because of backlash and toolpath not following perfectly. i will get the exact posts it really is only about 5 or six posts that have enough of the info to explain.
https://github.com/KevinOConnor/klipper/issues/3025#issuecomment-657906736 that was an earlier post by Kevin i think you saw the other ones from what i just read. 
[overlaptesttolerance_gage.3mf.txt](https://github.com/supermerill/SuperSlicer/files/4939057/overlaptesttolerance_gage.3mf.txt)
back right was best front was still good at .15 and the left was as well but was slightly oval. I am looking more into it. maybe if anything the important setting is just the one that moves the outer perimeter. I also am attempting the 0 overlap for infill and perimeter as you this klipper fork "well now the main branch with shaping on" really is huge improvement and is the first thing better than reprap i have found .  
[overlaptesttolerance_gage.3mf.txt](https://github.com/supermerill/SuperSlicer/files/4939089/overlaptesttolerance_gage.3mf.txt)

robthide37, 2020-07-17:
i think i am starting to understand more and really am not sure what the best solution is . obviously you know what your doing . I am curious why you default infill overlap to 15 percent and why it says no backlash would mean no need for that overlap. wouldn't it be the same ?

n8bot, 2020-07-17:
> i think i am starting to understand more and really am not sure what the best solution is . obviously you know what your doing . I am curious why you default infill overlap to 15 percent and why it says no backlash would mean no need for that overlap. wouldn't it be the same ?

It seems like several people are talking about different "overlap" features. There are several, if not many, scenarios where paths are overlapped in Slic3r/variants. SuperSlicer gives us the most control over them. In fact, I think you can remove any trace of overlap using SuperSlicer, but you can't remove the adjusted flow for extrudate shape (afaik).

robthide37, 2020-07-18:
I have been planning on sending 50 bucks to supermerril anyway just cause I use his software but I am at the point where I may just return this oscilloscope I just bought and donate the money to a few on here. My thinking is my tuning with motor specs and math using the tmc spreadsheet has improved spreadcycle to the point its as quiet as stealth chop at low speeds so no more low hanging fruit their and i may get a better return on orint quality by investing in few people on here
.
⁰

robthide37, 2020-07-18:
i only seee a way to do 2 or 5 a month how can i send a 1 time larger amount ?