supermerill/SuperSlicer #348: Continuous "outer" perimeter for thin walls,  also where inner structures intercept.

ruedli, 2020-07-17:
### Version
2.2.53

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S / MMU2S

### Behavior
I am printing wings for RC model airplanes. They use very thin (one perimeter) outlines, reinforce by "inner" rubs. All these structures are considered "outer"perimeters, as their surface they are on the outside of the 3D model. 

[image]

For finishing the "real" outer surface od the airfoil, I am interested in a real (continuously) printed outer contour. The Slicer does not generate these, or I haven't found the right setting. Where the "ribs" touch the surface, the contour is interrupted, see picture. I tried  things like: "activate outer layer first", "in vase mode (no seam)", deactivate "in "thin walls" "merging with perimeters", These had no effect.

Is there a way to influence the order in which the "outer" contour is printed? "Real" outer contour first and then inner ribs touching?, So _not_  like this, where the rib interupts the contour?

[image]

Fact: the Simplify 3D Gcode I received has the contour printed that way: the resulting surface is _much_ smoother.

#### Project File (.3MF) where problem occurs
[Elevon L1_.zip](https://github.com/supermerill/SuperSlicer/files/4937151/Elevon.L1_.zip)

robthide37, 2020-07-19:
try lowering line width maybe

supermerill, 2020-07-19:
it's not possible.
The wing shell is too thin to contain a loop, so it's extruded as a continuous single line.
The part where it join is able to be printed with loop, and as this is the preferred behavior, it's applied wherever possible.

You may be able to trick the software by separating your wing shell from the inner structure (where it's thick) by ~0.001 mm and disabling the 'slice gap closing radius'.

ruedli, 2020-07-20:
@robthide37 Tried lowering the line-width, went till 0.3, but it did not help.

ruedli, 2020-07-20:
@supermerill Thanks for looking into it. Unfortunately I cannot easily change the design: I bought the files and they provided .stl and GCODE done with Simplify3D. Your suggestion is very good, I feel it could have worked, but I do not want to try this without having access to the CAD files. As mentioned: Simplify3D probably uses a slightly different slice calculation algorithm: it has a continuous outer perimeter. Is it worth trying to slice it with Cura? I also tried ideaMaker: no success.

robthide37, 2020-07-26:
try checking the prusa slicer build at the end of #3025 issue in klipper. I
wonder if this could be a case of the outer perimeter overlap

n8bot, 2020-07-27:
This will not be affected whatsoever by the small change made in that build. The flow math does not make large scale changes to any behaviour.

closed 2021-04-27 by supermerill