supermerill/SuperSlicer #349: Missing slicing parts some models

BTopbas, 2020-07-17:
### Version
2.2.52

### Operating system type + version
MacOSx Catalina

### Behavior
On Superslicer some parts of models not slicing. Here is one of example. 

Superslicer:
[image]

PrusaSlicer or other slicers:

[image]

This is only one model I looked 3 different models but some parts not slicing and disappearing on the super slicer. I checked and unchecked thin walls and overlapping perimeters combinations then resliced but same happened. Maybe edit any other setting?

supermerill, 2020-07-17:
Duplicate of #335  #334 , #327 and #317
fix is in 2.2.52.1

BTopbas, 2020-07-17:
> Duplicate of #335 #334 , #327 and #317
> fix is in 2.2.52.1

Oh, my bad! I didn't look at the latest releases. Sorry about that! Closing issue.

closed 2020-07-17 by BTopbas