supermerill/SuperSlicer #350: Filament/Advanced/Print speed override does not seem to work

kursatu, 2020-07-17:
### Version
_Version of SupserSlicer used goes here_
2.2.52
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Printer: Custom corecy, Vking derivative
Firmware: Marling bugfix-2.0.x
### Behavior
* _Describe the problem_
The Filament/Advanced/Print speed override  setting does not seem to produce anything different.
* _Steps needed to reproduce the problem_
In the Filament/Advanced/Print speed override box enter a value, say 50
Slice
Observe that there is no gcode change in the file. The only change is in the metadata at the very end of the file.
Please see the meld diff attached.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Per the tooltip description, I've expected that M200 is generated for that filament wipe in between
; CP TOOLCHANGE WIPE
; CP TOOLCHANGE END
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_
[image]

[Mmu_test1.zip](https://github.com/supermerill/SuperSlicer/files/4939819/Mmu_test1.zip)

supermerill, 2020-07-17:
>  Filament/Advanced/Print speed

can't find it. Are you sure this thing exist?

kursatu, 2020-07-17:
It is again super useful feature.
There should a usable, very good speed control/settings for the wipe tower.
Currently, it seems to be fixed , depending on the filament type.
For me, it would decrease the filament switch time significantly.
I recall others complaining the petg not printing because the wipe speed it too high.

[image]

supermerill, 2020-09-09:
Okay, I axed my code when I merged with the new prusa wipe tower (new files, and I forgot to find a way to re-implement it, as the previous hooks doesn't exist anymore).

Also, the previous behavior of slower print with flex filament seems also axed. I'm trying things to re-create it.

kursatu, 2020-09-09:
Keep up the good work. 
Just a feedback, not sure what the behavior should be preferred. 
Prusa has chosen a fixed speed and slowing down via M220 for the wipe tower. 
My speed preference would be 
1- The speed defined in the filament settings
2- Optionally a speed percentage via "Print speed override"
3- The same speed should also apply to the empty grid
Currently , I am doing this (#3 and fixed speed up from #1)  via a post processor.
The time wasted at the wipe tower is pretty significant, especially when printing one-off , single copy models.

closed 2020-10-22 by supermerill