supermerill/SuperSlicer #351: Unnecessary, somewhat excessive travel moves while printing brim with multiple objects.

kursatu, 2020-07-18:
[Defect3.zip](https://github.com/supermerill/SuperSlicer/files/4941052/Defect3.zip)
[Defect3.zip](https://github.com/supermerill/SuperSlicer/files/4941053/Defect3.zip)

### Version
_Version of SupserSlicer used goes here_
2.5.2
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Printer: Custom corexy printer, Vking derivative
Firmware: Marling bugfix-2.0.x
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
Create multiple copies of an object that requires Multi-material printing
Slice with brim
Print and observe that the fist 2 brims makes extra travel move for almost each brim perimeter
It does not do that for the remaining brims.
Please see the image for the travel moves on the first layer, the right side has lots of extra travel moves.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Each brim is printed w/o any travel moves.
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_
[image]

yschroeder, 2020-07-23:
I also observed this. It basically prints the brim interleaved. One brim line for each brim and then cycles through them instead of finishing one brim and then moving to the next.

supermerill, 2020-07-23:
It's weird. I can replicate.
A bit difficult to debug, i don't know why it does that, and where the problem arise.

kursatu, 2020-07-24:
At some other models, it happens with all the brims. So it is more severe than initially reported.

closed 2020-10-22 by supermerill