supermerill/SuperSlicer #352: Implement continual solid (top) layers

ioma8, 2020-07-18:
### Version
2.2.52

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 MARLIN 2

### Behavior
Now top solid layers are printed in kinda random scattered order. Part of top infill is printed there, part there and this results in visible "joint lines" on tops of surface where single sub-surfaces are joint. Other slicers (eg e3d) print top infill in the most continuous way. Ie start at one corner and fill lines till the end, without skipping to other parts etc. Implement this in SuperSlicer pls.

_Is this a new feature request?_
yes

neophyl, 2020-07-18:
This is already being implemented in Prusa Slicer in an upcoming release. They call it Monotonous infill, and as such it will probably end up in Super Slicer eventually anyway.

ioma8, 2020-07-19:
Thx, you are right, I didnt know. From what I think, its already in their git repository? Thus couldnt it be already transfered to SuperSlicer?

neophyl, 2020-07-19:
I’m sure supermerill will correct me if I’m wrong but it probably makes more sense to wait until they have it all working and bug free before merging it I to the Super Slicer codestream.

supermerill, 2020-07-20:
It takes me ~2 days to merge their change last time.
And the last time i didn't merged their stable release, some bugs have also been merged & reported, so i'll wait something almost bug-free, at least an open beta.

ioma8, 2020-07-20:
Ok, makes sense. Thank you for all your efforts.

ioma8, 2020-07-22:
@supermerill hey I made the latest PrusaSlicer build from their master branch and that monotonous feature is great and makes me use only PrusaSlicer for now.

Would you mind if I forked SuperSlicer where I would merge this monotonous infill feature?

The latest windows PrusaSlicer build I made is here if you wanted try: https://drive.google.com/uc?id=1zKA2uFxJSYlYoBZlcOi5yS8BsOhL_pWF&export=download

supermerill, 2020-07-23:
It's agpl open-source, you can do whatever you want as long as you credit.

ioma8, 2020-08-24:
> Good thing: less release, is less merge, so more time to do other things.
> Bad thing: less release mean more difficult merge, and they advertised
> "alpha feature" months before their release...
>
> @ioma8 <https://github.com/ioma8> if you succeed to merge the monotonous
> infill, you can send a pull request.

Hi yes, Im on vacation now but could look at it in 14 days cca. So far I
have been using the monotonous feature in the PrusaSlicer built from their
master branch and it is realy great, I do not wanna come back to normal
infill. :)

closed 2020-10-22 by supermerill