supermerill/SuperSlicer #353: Makes manilfold when not supposed to be and deleted first layer

robthide37, 2020-07-19:
it is reversing the main body so that it prints in mid air
[mouse_trap_main_body_no_support.3mf.txt](https://github.com/supermerill/SuperSlicer/files/4943465/mouse_trap_main_body_no_support.3mf.txt)

### Version
_Version of SupserSlicer used goes here_

_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

robthide37, 2020-07-19:
i cut bottom half mil off to fix but i am curious why it renders properly but then slices inverted for first layer. If you rotate on back or upside down it once again slices properly

supermerill, 2020-07-20:
[image]

what/where is problem?

supermerill, 2020-07-20:
imo, it's a bad model. cut at ~0.1mm and keep only the top part.

neophyl, 2020-07-20:
Definitely a damaged model.  

I imported the stl into blender and ran its 3d print check tools which showed, 113 Non Manifold edges, 296 intersecting faces, 53 zero faces, 2 zero edges and 293 thin faces.  After using the clean up / make manifold button it still showed a few issues (5 non manifold edges, 221 intersecting faces etc) and re imported it back into Super Slicer.  The fixed version looked like it slices fine whereas the original add in the solid section on the bottom which lifts the model up.  The fixed version also doesnt show the auto repaired 52 errors that the original has.

closed 2020-07-23 by robthide37