supermerill/SuperSlicer #356: Adding SuperSlicer features in a "Super" tab

Mithrandil, 2020-07-21:
NEW feature request:

Moving SuperSlicer only features (compared with PrusaSlicer) in a "Super" tab:

[image]

supermerill, 2020-07-21:
lol

why?

neophyl, 2020-07-21:
You can already do that if you want can’t you ?  
https://github.com/supermerill/SuperSlicer/tree/master/resources/ui_layout

n8bot, 2020-07-21:
Make it easier for new users to discover new features and distinguish what's changed.

I think it should be one level above "Expert," a blue icon: SUPER!

supermerill, 2020-07-21:
> Make it easier for new users to discover new features

I prefer making a wiki & videos. If I can find the time.

n8bot, 2020-07-21:
I would volunteer to help with that, if you need any help. Just let me know, I'm sure you know where to find me. lol

ioma8, 2020-07-22:
@supermerill hey I am open to helping with wiki too.

Mithrandil, 2020-08-26:
It seems great!

Just pay attention not to make it a pain to mantain when updating to new PrusaSlicer versions

supermerill, 2022-02-21:
In the latest nightly, you have it.
I'll hide it for the release, but you can re-show it by uncommenting the line in the resources/ui_layout/colors.ini
I'll try to create a kind of "UI module" so you can save & share UI layout more easily.

closed 2022-07-14 by supermerill