supermerill/SuperSlicer #358: Flow Calibration doesn't finish slicing or has empty preview with a certain filament profile

paul-arnold, 2020-07-22:
### Version
2.2.52

### Operating system type + version
Win 10 Pro 

### 3D printer brand / version + firmware version (if known)
P3Steel Self built, running Klipper (latest)

### Behavior
I tried to calibrate a new spool today.
However, the 5 flow calibration objects (-20 - +20) wont finish slicing or dont show a preview with a certain filament profile .
Any other filament profile works good.
It is reproducable. 
Screenshot:
[image]

[Supermerrill_issue.zip](https://github.com/supermerill/SuperSlicer/files/4962454/Supermerrill_issue.zip)

#### Project File (.3MF) where problem occurs
-> attached with filament profile

supermerill, 2020-07-23:
an extrusion multiplier of 0?
I should try to add a safeguard on this setting.

paul-arnold, 2020-07-25:
> an extrusion multiplier of 0?
> I should try to add a safeguard on this setting.

Yikes, indeed. How could I not see that?
Setting it back to 1 fixed the issue.
I probably tried to set it to 0.something and the dot plus "something" was lost somewhere.
Thanks for helping out.

closed 2020-07-25 by paul-arnold