supermerill/SuperSlicer #359: Repetier server

den180276, 2020-07-22:
### Version
2.2.52

### Operating system type + version
Mac OS High Sierra

### 3D printer brand / version + firmware version (if known)
VORON 2.2 Klipper

### Behavior
no support Repetier Server Host

supermerill, 2020-07-23:
ok.

I guess you post this not only to say me that. So maybe it's a "please, can you add a support for repetier" ?
If it's the case, please help me, and tell me what i need to do. I don't use repetier and i don't have the time to learn what is it and how it works.
Do you have links to wiki? Or link to other similar issues in the prusaslicer/slic3r repository?

den180276, 2020-07-25:
Hi! 
I may have described the problem incorrectly. Can you add repetier server support ? I don't have links to the Wiki, but repetier is the same octoprint. To connect, you need an IP and API key.

Lukas1818, 2020-07-27:
There is an [example script](https://www.repetier-server.com/using-simplify-3d-repetier-server/) for uploading and starting the print, which looks very simple. The full documentation of the api can be found [here](https://www.repetier-server.com/manuals/programming/API/index.html). Because of the multi printer support of repetier server you need to use the printer-name as addition parameter. It should be possible to test the program with the [repetier server live demo](http://demo.repetier-server.com:4005/)

You may want to take a look at [this](https://github.com/prusa3d/PrusaSlicer/pull/4384) merge request.

supermerill, 2020-07-31:
I'll look if i can pull the merge request.

closed 2020-11-25 by supermerill