supermerill/SuperSlicer #361: "Show incompatible print and filament presets:" -checkbox does not work

jacksterz, 2020-07-24:
### Version
2.2.52

### Operating system type + version
Windows 10 Enterprise

### 3D printer brand / version + firmware version (if known)
Voron V2.2, newest Klipper firmware

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  In Preferences, check the "Show incompatible print and filament presets:" -checkbox
* _Expected Results_
  Incompatible filament profiles should be shown in "Filament Settings"
* _Actual Results_
  Incompatible profiles remain hidden

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

neophyl, 2020-07-24:
It does work, it just doesn’t work like you would expect it to. Before Prusa implemented the new inheritance scheme it used to work that way, now the new way is the primary way and the show incompatible profiles only works within that vendor group.  

The filament profiles are not incompatible in the sense that the option covers, they are locked to the Vendor which for most of them is defined as Prusa. In Prusa thinking it’s because they have stuff like linear advance k factor as part of them which means they are only suitable for their printers. I had this conversation with one of the devs about it as I strongly disagreed on locking out the filaments for other printers. They are needed as a starting point for others to customise for their non Prusa printers. He didn’t agree but did say they would discuss it internally. 

The only thing you can do is add a Prusa printer to get access to the full list and then create copies of the ones you want while remembering to click the detach from system preset button in the filament profile before saving it again. 
Or you could just install the filaments from the following archive I posted on the Prusa forum to save you some work. https://forum.prusaprinters.org/wp-content/uploads/2020/04/GenericFilaments_Neo_20200401.zip
Basically I took most of the filaments, copied them, detached them all and saved them into a config bundle. You can unzip and then import the bundle which should get you the majority of them unlocked for any printer. As the k factors etc are still all in there use at your own risk.

jacksterz, 2020-07-24:
Ok, thank you for the clarification and the workaround!

closed 2020-07-24 by jacksterz