supermerill/SuperSlicer #362: Ironing speed?

joeynuggetz, 2020-07-24:
Is there a way to set the ironing speed? Is there a max recommended speed? Would setting top layer speed accomplish this? Also would it be possible to iron any layer to provide a nice base to print for the subsequent layers? I think some high end printer iron each layer. Would like to experiment. 

Also I’ve tried 10, 15, and 20 percent flow with overlap down to 25% and results are terrible. Prusa MK2.5s.

supermerill, 2020-07-25:
> Is there a way to set the ironing speed?

change the top infill speed

> Is there a max recommended speed? 

~ 40

> Would setting top layer speed accomplish this? 

yes

> Also would it be possible to iron any layer to provide a nice base to print for the subsequent layers? Would like to experiment.

it's not needed. If you really want to experiment with it, change the pattern in the "solid pattern" setting to ironing.

> Also I’ve tried 10, 15, and 20 percent flow with overlap down to 25% and results are terrible. Prusa MK2.5s.

Don't change default settings. They are good. If you have bad result with pla, check the tooltips and [the wiki](https://github.com/supermerill/SuperSlicer/wiki/ironing)

closed 2021-04-27 by supermerill

marcosscriven, 2021-08-02:
> it's not needed. If you really want to experiment with it, change the pattern in the "solid pattern" setting to ironing.

I’ve just been experimenting with ironing “all solid surfaces” in the 2.4 alpha branch of PrusaSlicer. The GUI option has been there a while, but the final implementation in ‘fill.cpp’ is only in the latest alpha tag. 

Curious how you’re planning on merging these two approaches?

supermerill, 2021-08-02:
> the final implementation in ‘fill.cpp’ is only in the latest alpha tag.

It's merged in the master since months. and in all 2.3.55 & 2.3.56 releases.

> Curious how you’re planning on merging these two approaches?

Currently it's not needed to merge them. You can use the one you prefer, or even both (to do a triple-pass ironing).
As the prusa one is completely separate from the pattern, and over-extrude on top of them, there is no real need to do something.

I'm also considering to use some of the new setting like 'ironing speed' instead of 'top speed'... If it's more intuitive and won't create issues.