supermerill/SuperSlicer #363: Dual extruder issue/question

Tinchus2009, 2020-07-24:
### Version
Latest

D`_

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
Custom buil, dual extruder

### Behavior
Hello. I have a custom printer with a dual hotend configuration. Im using a duet 3 with reprapfirmware. Usually, Im using cura, and now im trying this superslicer and it looks really good.
The only problem Im having now is with dual printing: is it possible to have a wipe tower? 
Because I set extruder 1 for the main material and I want to use extruder 2 for soluble support material. But when I try to configure it like so, I receive a message saying that "the wipe tower currently supports  the non-soluble support only if they are printed with the current extruder..."
Im doing somehting wrong or the slicer doesnt support dual extrusion systems and wipe towers together?
Is that feature a possible addon in a future?

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-07-24:
in "multiple extruders" tab, set the two support extruders to 0.

n8bot, 2020-07-25:
Can you elaborate on what the extruder 0 is for and how it helps in this situation?

n8bot, 2020-07-25:
I found my answer in the codebase:

>// For non-soluble supports, the default "0" extruder means to use the active extruder,

supermerill, 2020-07-25:
it's also explain in the tooltip

Tinchus2009, 2020-07-26:
I saw the tooltip message, but that is not answering my question.
I have 2 extruders, in one I use ABS, in the other one I use PVA. PVA was set as SOLUBLE.
SO my intention was to setup superslicer to use ABS on extruder 1, and soluble support on extruder 2 and have a wipe tower. When I try to set up this I keep receiving a message that I have to set up the extruder 2 to a value of CERO. If I accept that, everything is printed with extruder 1 and extruder 2 is not used.
If I say NO, extruder 2 is used but no wipe tower is used.
There are 2 important problems with this:

1.- when using PVA, you cant have PVA at printing temperature withoit being extruder sitting on the hotend, it will degrade very quickly. PVA build carbon deposits very fast if you do that. So you need the OOZE PREVENTION feature. IT is mandatory to educe temperature in these situations. But the frature is not compatible with wipe tower.
2.- if you use the ooze prevention feature, a tall skirt is made, but it doesnt work really well, because that tall skirts doesnt have a brim. Building such a skirt with ABS is imposible without getting a warped skirt after some layers.

Im a fan os lic3r since a lot of time, I dropped it when I saw the limitation in dual extrusion. The aprouch of superlsicer is really great, I can see a lot of powerfull extra features added. But would be nice to see more powerfull features for dual extruder printers.

supermerill, 2020-07-31:
Ooze shield with purge block: it's possible, see #371

brim for ooze shield: just put a higher value in brim width than in skirt distance from part.

closed 2021-04-27 by supermerill