supermerill/SuperSlicer #364: "Estimated printing time" incorrect

ruedli, 2020-07-24:
### Version
2.2.52

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s / MMU2S

### Behavior
When slicing the printing estimates are incorreect. Also the esitmates per layer (as displayed in Octoprint) are wrong.

I am using a MMU2 profile, and attached the project file with settings. All my MMU2 prints show this. I am using **height modifiers** to specify extruder transitions, this could be contributing.

In the standard Prusa Slicer, the problem does not occur.
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
[cover_lcd_skr.zip](https://github.com/supermerill/SuperSlicer/files/4974459/cover_lcd_skr.zip)

codingcatgirl, 2020-07-25:
This happens in PrusaSlicer too. 
For me it always seemed to have something to do with brim size, but in general it seemed to be almost completely unreproducible sometimes.

ruedli, 2020-07-25:
I could bring the same objects in again, and they sliced without problems now... Very strange with its reproduceability... Repairing the non-manifold element was not influencing.

supermerill, 2020-07-25:
I think it's random.
maybe next prusa release will correct it.

robthide37, 2020-07-26:
yes it is you can change the infill when it happens and it will self correct. i think it happens more when background slicing is on and it tries to recycle some calcluations. you would know better but seems some reslices are way faster so something caches and when you change a setting it messes things up with the time estimation

Alonso3D, 2020-07-27:
Ive had this issue as well. For me it would happen when usingbthe wipe tower. See #276

vertigo235, 2020-08-24:
Same issue here, didn't realize it was in Prusa Slicer too (ditched that pony a while back!)

Alonso3D, 2020-08-24:
I'll add that after purging dual nozzles in the beginning, the gcode tried to move the nozzle to -X (rather huge number) causing the printer to shutdown. I had to manually adjust the gcode to something sane like X 50 and it worked fine. I beleive the massive move is what is contributing to the crazy time estimates

stelgenhof, 2020-08-30:
The infamous -2^31 :) Likely an overflow condition or missing value somewhere.

supermerill, 2020-09-19:
I've added a check, that print a message on the console and retry to compute the printing time.

Lukas1818, 2020-10-14:
I can confirm that the issue still exist in SuperSlicer 2.2.53.3.

Lukas1818, 2020-10-15:
The time calculation seem to be correct at the latest version (de57391b7966c0fddb15a93f2c6abc393a03e21c). I was not able to reproduce the issue with this version.

Lukas1818, 2020-11-07:
I get this issues in Superslicer  2.2.54.2 again.

closed 2021-06-04 by supermerill

codingcatgirl, 2021-06-04:
Oh, is this gone? That would be awesome news! Haven't seen it in a while.