supermerill/SuperSlicer #365: Ironing coming up perfect, but chewing up filament, thus stopping extrudes

nfons, 2020-07-25:
### Version
_Version of SupserSlicer used goes here_
2.2.52.2

The ironing is coming up **PERFECT**.  smooth as a injection mold..HOWEVER, during large prints, the filament seems to be getting chewed up by the extruder in parts of ironing

I have done flow calibration, and its near perfect (0.78-79 on 0.8mm cube)

The above bridge flow rate is set to 120%

Top solid 0.4

i have tried 50% width @ 10% distribution (default)

As well as 50% @ 20% distribution.

in all of these most parts of ironing will be perfectly fine, then stop extrudes due to filament chew. so only something like 75% will be ironed.

Printer: Prusa mk3s.

is this due to settings or the printer itself?

puneit-hassio, 2020-07-25:
I have the same issue as well. I am thinking of disabling ironing.

nfons, 2020-07-25:
I really do want to re-iterate that the ironing itself is PERFECT. like _I cant believe my eyes_  perfect in terms of how smooth it is.

it just so happens that it will crap out half way through the process in large prints

puneit-hassio, 2020-07-25:
Yes I am impressed (super impressed) by ironing- I assume we have the same issue here- on large prints the extruder eats away the filament and then the printer stops extruding. 
The filament inside the eat ends keeps cooking and swells up. 

I am on ender 3 pro, direct drive mod, skr mini e3 board. Stock extruder and hotend.

supermerill, 2020-07-31:
2 things to verify:

You may want to limit the power you put into the extruder. It need to stay cool.
If you ask to extrude while the heater is not hot enough, does it chew your filament or does it click (miss steps); If it chew it instead of missing steps, try to lower the power it get a bit.
If the motor is very warm, you may need a radiator and a fan, or a belt to not transmit the heat to the filament.

Also, you may need to tune your extruder tension. It needs to be tight enough that you can't pull the filament by hand (the motor should turn if power down, or the filament has to break/snap if the motor is powered up). But use the least amount of tension possible.

Pressure advance may cause some issues on the ironing passes (too many retractions on the same part of the filament). If you have a very big PA value, and has this "chewing up filament" issue, please add a comment below tell me that, as i may have to add command to remove PA during the ironing phase.

Also, their is currently no pressure system i know of in this slicer nor in firmware. The pressure needed depends on the flow, but i don't think anything takes that into account.

nfons, 2020-08-01:
@supermerill Thanks for the response.

the tension seems to be as low as it can be.

The extruder stepper is might be hotter than it should be (sits @ ~50C). I will try adding some heat dissipate

for the other things, PA = Linear Advance? 

voltage is what ever the default voltage is in 3.9.0 fw.

supermerill, 2020-08-01:
The extruder assembly of the prusa mk3 give too much heat into the extruder gears, for what i can read on the social medias. It's not a problem unless you extrude very little amount of plastic, like for the ironing, because the pla will heat up too much.
Not much i can do on the slicer side.

nfons, 2020-08-02:
thanks, Let me try to add some heatbreaks to the extruder to help dissipate the heat. will check back on it after

totalitarian, 2020-08-02:
Try ironing at a faster speed. You need a certain flow to prevent clogging

nfons, 2020-08-03:
> Try ironing at a faster speed. You need a certain flow to prevent clogging

@totalitarian clogging isnt happening. the filament is being chewed from what i think is repeat retracts.

if during these "stoppages" i manually push the filament to bypass the chewed section, it will work.

nfons, 2020-08-05:
@supermerill I have eliminated extruder heat recently, and set distribution @ 15. 

This worked fine for smaller prints, but larger ones same issue again.

supermerill, 2020-08-05:
Can you test with 0 retraction?
If it works, I'll see if I can add a "no retraction on top infill" modifier.

totalitarian, 2020-08-05:
Im still convinced the pla cooks and clogs in the hotend causing the
extruded to grind the filament.

nfons, 2020-08-05:
@totalitarian we can easily test that and disprove. (which i did test).

if the pla is clogged and is causing the filament to grind, then adding "new" filament would also grind the PLA.

I have verified that adding new filament :

1. Either as a simple manual push to get past the grounded up section
2. Brand new filament itself

the extruder will work as intended.

Furthermore, we can eliminate nozzle clogs by understanding that if nozzle clogs at temp **X**, the only way to clean the nozzle w/o a manual intervention such as needle poking  would be to heat the nozzle significantly past **X**.

But since we can get nozzle to extrude @ X temp, we know its not a nozzle clog.

@supermerill I will try to do some new prints at large size with some modifiers to eliminate retracts at the top end and see if that solves it.

nfons, 2020-08-08:
Ok So in order to ensure everything is as sane as I can possibly make it:

- Eliminated Heat cream to extruder via heatsink. Extruder is just a little above room temp.

- Modified iron distribution as well as width to various degrees.

- modified flow.

none of these seemed to have helped. Filament is still chewed up.

disabling retract DID help though (but for obvious reasons, the model sucked due to stringing)

WemansG90, 2020-08-30:
Going to jump in here: Ironing larger areas seems to be clogging/messing with my PLA prints as well. Heavily modded Ender 3 with an all metal hot-end, 30:1 dual gear(BMG) extruder and a duet maestro. 

It's similar to the clogs I used to have when my retraction distance was set too high, needing the clog to be "cooked" out at 235°C or so.

nfons, 2020-08-31:
> Going to jump in here: Ironing larger areas seems to be clogging/messing with my PLA prints as well. Heavily modded Ender 3 with an all metal hot-end, 30:1 dual gear(BMG) extruder and a duet [...]

Did you check the filament near extruder gear? is it chewed up?

WemansG90, 2020-08-31:
@nfons Yeah, once it clogs, the filament starts slipping/the stepper really complains when I crank down the gears. But I think the slippage is due to swelling causing a block, not the other way around

supermerill, 2020-08-31:
Too many retractions on my ender-3 also create a small blockage.
It's maybe due to overheating of the filament in the heatbreak area, too many time spent here when retracted.

Imo, the right answer may be a retraction watchdog to force smaller/no retraction when there are too many of them.

madsi1m, 2020-09-20:
I found after ironing I suffer from a little of under extrustion, eg: https://www.thingiverse.com/thing:1278865 after ironing the surface of the letter Z, I get 2 layers of under extrustion, then it goes good again.

Perhaps i have too much ooze during ironing?

supermerill, 2020-09-20:
It's more of a lack of pressure model.
When you have a low flow, you don't need much pressure.
When you're going fast and big, you need a lot of pressure.
The change between these extrusions isn't taken into account, you should have an unretraction to go from low to high, and a retraction to go from high to low (flow).
Pressure advance may help but i'm not sure by how much.

hagan66, 2021-06-01:
Hi! I also have this problem. The ironing results are great, but filament gets chewed and stops extruding. I have a Prusa MK2.5 on which the extruder motor never gets hot. Not even warm. I havent come up with a workaround other than disabling ironing.

rananna, 2021-12-31:
I am also having this isssue.
No retraction on top infill would be a great option to try out.
Could you consider implementing this?