supermerill/SuperSlicer #366: Crashing when enabaling sequential printing and/or changing sequential parameters

darrenph1, 2020-07-26:
### Version
2.2.52

### Operating system type + version
Mac OS 10.13.6

### 3D printer brand / version + firmware version (if known)
MK3s / Vertigo 3.9.1++++ LA 1.5

### Behavior
Trying to do sequential printing and it crashes a lot when changing selections
* _Steps needed to reproduce the problem_ 
If I set up 2 objects I get the warning that the objects are too tall. I try and change the object sort and it crashes.

It has also crashed randomly when I just enable sequential printing
[Seq_config.ini.zip](https://github.com/supermerill/SuperSlicer/files/4976949/Seq_config.ini.zip)

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_
No
[Sequential mode crash.3mf.zip](https://github.com/supermerill/SuperSlicer/files/4976947/Sequential.mode.crash.3mf.zip)

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

supermerill, 2020-07-31:
I can't reproduce the crash, maybe i already fixed it on my master branch.

it's not possible to print them sequentially with your height clearance anyway.

neophyl, 2020-07-31:
I tried this with the 2.2.52 release and it wouldnt crash on me either.   (Windows 10 64 bit).

supermerill, 2020-07-31:
As you're on macos, maybe it's the red warning that make it crash.

Try to add a object into the plater, then move it outside of the plater. Do you see the red warning or does it crash?

darrenph1, 2020-07-31:
> As you're on macos, maybe it's the red warning that makes it crash.
> 
> Try to add a object into the plater, then move it outside of the plater. Do you see the red warning or does it crash?

That's it. I can add 2 smaller items and it's fine. Soon as I overload the plate and enable it it crashes, no red warning.

puneit-hassio, 2020-07-31:
I have seen the red warning on macOS. The crash happens for me as well and
most of the times , application crashes when generating calibration
objects. Otherwise , crash is extremely rare.

supermerill, 2021-06-26:
I think it's fixed now.

closed 2021-06-26 by supermerill