supermerill/SuperSlicer #368: multithreading

Coffee0297, 2020-07-27:
### Version
All

### Operating system type + version
All
### 3D printer brand / version + firmware version (if known)
All

### Feature request
[image]
Is it possible to make the slicing multithreaded? and the gcode generation? its real slow when enabeling features that is clearly single threaded.

supermerill, 2020-07-31:
> Is it possible to make the slicing multithreaded? 
it is where possible

> and the gcode generation? its real slow when enabeling features that is clearly single threaded.

yes, it's not possible to multi-thread a sequential process like writing the gcode file, as you need the result of the previous move to compute the next one.

Coffee0297, 2020-07-31:
So it wuld not be possible to "slice" the model and assigning parts of the model to difrent threads? and just use 1 core to "merge" them? im asking bc slicing a high res model it takes like 10-30min just for the gcode generation.

Kind Regards Tonny

supermerill, 2020-07-31:
problem is you can't know where to start in the "part" if you don't know where the last extruder position is, or you'll have absurd travels.

supermerill, 2020-07-31:
You can send me your project, and i may be able to maybe speed it up a bit if i found a bottleneck that can be optimised.

Coffee0297, 2020-07-31:
yea i just wait on it completes, i was just wondering if it was possible

Kind regards Tonny

closed 2020-08-04 by Coffee0297