supermerill/SuperSlicer #369: Hotend pauses while printing on printed object

DasBootU96, 2020-07-27:
### Version
Version: 2.2.52
Build: SuperSlicer_2.2.52+UNKNOWN

### Operating system type + version
Windows Version: Windows 10 (build 18363), 64-bit edition

### 3D printer brand / version + firmware version (if known)
VoronDesign Voron V2.2 DIY CoreXY
FIRMWARE_VERSION:v0.8.0-426-g8ac81fed FIRMWARE_NAME:Klipper

### Behavior
* Hotend pauses for 2 seconds (sometimes 5 seconds) while printing fine details causing blobs
* _Steps needed to reproduce the problem_
  * Download and slice Nano 3D Print Test from https://www.thingiverse.com/thing:2887065
* _Expected Results_
No hotend pausing on the printed object anytime while printing
* _Actual Results_
Pausing for 2 seconds on some of the 3D print test fine objects. Not an issue when PrusaSlicer is used.
  * Screenshot from __*SuperSlicer*__ preview:
[image]
  * Print from __*PrusaSlicer*__ vs __*SuperSlicer*__:
[image]

Used ABS+ from eSun and yes, I'm underextruding.

_Is this a new feature request?_
Nope, feels like a bug.

#### Project File (.3MF) where problem occurs
[Nano_3D_Print_Test.3mf.zip](https://github.com/supermerill/SuperSlicer/files/4984228/Nano_3D_Print_Test.3mf.zip)

supermerill, 2020-07-31:
It's a problem of overloading the connection between octoprint and the firmware.
I've worked on it, but it's not live yet.

DasBootU96, 2020-08-25:
Tested with SuperSlicer_2.2.53.0 and the pausing is still there albeit slightly less so. Attached example photo from print sliced in SuperSlicer_2.2.52.1 vs SuperSlicer_2.2.53.0.

Is there anything else to try?

[image]

supermerill, 2020-08-25:
In printer -> general -> minimum extrusion length, there should be 0.035
can you test with a higher value like 0.1 ?

DasBootU96, 2020-08-25:
The 'minimum extrusion length' was indeed set to 0.035, changed it to 0.1. Still pauses, see image below.

Perfgraph from Klippy.log shows a host buffer at 100%. FWIW: Using a Pi4 and SKR1.3s (2209 drivers).

Should I go higher with 'minimum extrusion length' or try something else?

[image]

[image]

DasBootU96, 2020-08-26:
Removed most octopi plugins and print completes without pausing. Now adding plugins back 1 by 1 to see if I can identify the culprit. Will follow up soon(ish).

DasBootU96, 2020-08-26:
After installing '[DisplayLayerProgress Plugin (1 23 0)](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/)', the pausing was back though not in the exact same pattern as before. But it seems safe to say this plugin is a major cause. It was installed to enhance the '[OctoPrint-Dashboard](https://github.com/StefanCohen/OctoPrint-Dashboard)' plugin.

Will try OctoPrint-Dashboard next to see if the exact same pattern returns.

[image]
[image]

closed 2020-10-22 by supermerill