supermerill/SuperSlicer #370: Error when debugging

jorksx, 2020-07-27:
Hello. I followed "noob steps" from here. https://github.com/supermerill/SuperSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md
everything went perfectly BUT, when I start debugging or start without debugging, console opens with text: Slic3r.dll was not loaded, error code: 126
What did I do wrong?

[image]

Pavulon87, 2020-07-30:
I'm not sure, but you could try:
Build -> Build Solution [CTRL + SHIFT + B]

supermerill, 2020-07-31:
be sure that in your folder build/src/Debug you have libgmp-10.dll, libmpfr-4.dll and superslicer.dll
for the first two, you can copy them from a release
the last one should be generated when you compile

kursatu, 2020-08-01:
I was running into the same issue.
Alternatively, build from the command line using, which fixed it for me.
The missing 2 dll's are built using this command line
SuperSlicer\build>msbuild /m /P:Configuration=Debug ALL_BUILD.vcxproj

jorksx, 2020-08-02:
> be sure that in your folder build/src/Debug you have libgmp-10.dll, libmpfr-4.dll and superslicer.dll
> for the first two, you can copy them from a release
> the last one should be generated when you compile

first 2 files are there. superslicer.dll is still missing even after compile.

supermerill, 2020-08-02:
> superslicer.dll is still missing even after compile.

check the compile log to see where it's written, or via a file explorer search.

jorksx, 2020-08-02:
I went easier route and downloaded pre-build one. So problem fixxed for now :)

closed 2020-08-02 by jorksx

n8bot, 2020-08-02:
If you are debugging changes you've made, you'll certainly want to make sure the superslicer.dll is the one you build yourself.