supermerill/SuperSlicer #371: Feature Request: Ooze Shield for Multiextruder prints

lukeslaboratory, 2020-07-27:
Hi! - I have a large toolchanger build and really love Superslicer.

However - a feature that I really like using in terms of consistency in priming the extruder and preventing dribbles is the ooze shield from Simplify3d - https://twitter.com/i/status/1278313756385333251 -

Its a shell fitting  that primes the tool before going into the main print. it has a nice effect of catching any ooze that leaks from the toolchange post and doesn't give much space before it goes into the tool.

I currently use a purge block with superslicer, but would love it if the feature could be added.

Thanks for reading!

supermerill, 2020-07-31:
you can't use a tall skirt ?

lukeslaboratory, 2020-07-31:
Unless I'm dumb, tall skirts only use one extruder. I'd like to have a skirt that uses all active extruders on a multi-material part.

lukeslaboratory, 2020-07-31:
Confirmed, Am dumb. Had to swap to "0" for skirt extruder, and add at least as many loops as tools used, Priming skirt enabled.  tighter-fitting shell still applies

supermerill, 2020-07-31:
> tighter-fitting shell still applies

It's difficult to code, as you can't have a big overhang. Big feature.