supermerill/SuperSlicer #372: [Bug] Variable layer height sometimes not working

Pavulon87, 2020-07-28:
### Version
SuperSlicer_2.2.52.0_win64_20200623
SuperSlicer_2.2.52.1_win64_20200703

### Operating system type + version
Windows 10

### Behavior
Since the 2.2.52 "Variable layer height" sometimes don't work as it should. May be related to "Z full step"(set to 0.04).

Project from 2.2.51 imported in 2.2.52 lose variable layer height settings/profile, but after applying it is ok.

2.2.51 - working:
[image]
**2.2.52 - not working:**
[image]
2.2.52 - preview:
[image]

2.2.51 - working:
[image]
**2.2.52 - not working:**
[image]
2.2.52 - preview:
[image]
2.2.52 - working accidentally
[image]

2.2.51 - working:
[image]
2.2.52 - working:
[image]

2.2.51 - working:
[image]
2.2.52 - working:
[image]

#### Project File (.3MF) where problem occurs
2.2.52 - [Body1(1).3mf](https://github.com/supermerill/SuperSlicer/files/4986238/Body1.1.zip)
2.2.51 - [Body1(2).3mf](https://github.com/supermerill/SuperSlicer/files/4986239/Body1.2.zip)

kursatu, 2020-07-30:
It is an essential feature.
Any possibility that it can this be fixed soon?
I have observed this as well. 
I've the "Enable variable layer height" enabled in the printer setting, but still does not work.
Not sure if it ever works or ever worked.
@Pavulon87 Where is that "Z full steps" setting?
Or is there any pattern for it working accidentally?

neophyl, 2020-07-30:
I loaded in both your body 1(1) and body 1(2) projects into 2.2.52,  I can confirm that (1) the variable layer settings are loaded with the project and that with (2) the variable layer heights have been reset and are not there (I assume they are supposed to be ?).

One difference I did notice between the projects though is that in (1) the Z stepping is set to your 0.04 value that you mentioned but (2) the stepping value is set at 0.005.  

The Z stepping value isnt being saved in (2) naturally as this project was saved by 2.2.51 which doesnt have the Z stepping value. I suspect when its imported in to 2.2.52 it gets set to the default of 0.005 which in turn invalidates the variable layer settings so you have to do them again but once generated it works ok as the steps match what is used for stepping when generated.  

If you look inside the 3mf files in the Slic3r_PE.config files then the (1) contains at the very bottom ; z_step = 0.04 but the same file for (2) does not have that configuration entry.  Doing a compare on the 2 config files there are several more entries that the one saved in 2.2.52 has over the one saved by 2.2.51.  This isnt surprising as 2.2.52 has more settings.  

So in practice it looks like variable layer maps are not compatible between projects saved with 2.2.51 and 2.2.52 at the moment.

Pavulon87, 2020-07-30:
It seems that some values are working and some are not.
0.01 - OK
0.02 - not OK
0.03 - OK
0.04 - not OK
0.05 - not OK
0.06 - not OK

But after resizing model height to round millimeters(18.51mm ->19mm) there is everything OK.

Pavulon87, 2020-07-30:
After changing default value in source files it works great. Legend is correct either.
def->set_default_value(new ConfigOptionFloat(0.04));
[image]

I'm sure that default value is used instead of config where it should not, but i'm too dumb to figure out where is the problem.

kursatu, 2020-07-30:
The problem happen w/o a project file as well.
That what it makes this feature unusable.
I did not find a workaround yet. Changing the Z full steps to the default value did work as a workaround.
Please see the attached images.

[image]
[image]

Pavulon87, 2020-07-31:
ohh... Plater.cpp 
Works great now.

closed 2020-07-31 by Pavulon87