supermerill/SuperSlicer #374: Ironing used for first flow calibration test

dkatzdev, 2020-07-28:
### Version
2.2.52.1

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S / 3.9.1 (patched)

### Behavior
The calibration test for filament flow multiplier currently uses ironing for the top layer. This adds other variables to which sample prints the best, as poor ironing settings can misleadingly cause poor top surfaces.

Changing the test to generate objects with rectilinear tops would likely solve this by reducing the amount of variability in other settings and result in the filament flow being the main factor of surface smoothness.

supermerill, 2020-07-31:
I'll do some tests to verify if it's better.

dkatzdev, 2020-07-31:
I had a massive issue tuning my extrusion with PETG since the ironing made the top surface really rough on anything other than heavy overextrusion, leading me to use a multiplier of 1.08, giving me prints with bad tolerance issues and overextrusion. By turning off ironing, I got really smooth tops and found my actual flow multiplier of 0.94 which is much more dimensionally perfect without any underextrusion

supermerill, 2020-07-31:
i'll do some wide test prints and see how much they weight vs the computed value.
I think you should be right, the ironing may deplete a bit too much pressure as their are no mechanism to regulate it (yet).

yschroeder, 2020-08-01:
On my mostly stock ender 3 I also got very rough top surfaces with the default ironing settings, if this information is of any help.