supermerill/SuperSlicer #377: First_extruder placeholder

Exerqtor, 2020-08-01:
### Version
2.2.52

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Home made CoreXY / Duet 2 running RRF 2.05.1 and DWC 3.1.1

### Behavior
Would it be possible to add a "first_extruder" placeholder for start g-code, that simply puts the extruder number to be used for the initial part of the first layer? 

It would be super helpfull for us with multi-extruder single nozzle printers so we wouldn't have to swap around filaments on the printer etc so often.

For example if my last print finished of with lets say T2 who was loaded with CF-nylon filament, and my next print is to start with T1 and who's loaded with PLA. I would want to have purge the tool from nylon in the startup by having a hugh purge temp using "first_extruder" to deploying T1 so that it to run and prime the nozzle with the filament actually going to be used after the start g-code.

supermerill, 2020-08-04:
you can use initial_extruder or current_extruder

Exerqtor, 2020-08-04:
Oh ok, couldn't find anything about that in any documentation. Thanks!

closed 2020-08-04 by Exerqtor

supermerill, 2020-08-04:
np, the doc is a bit out-of date.

Exerqtor, 2020-08-04:
Yeah all the slic3r documentation is way out-dated for all forks, on that accord which of the files for contain the different condition's/variables that's actually available to use for macros etc (or are they spread out all over)?  So one could have a look at it and at least see what we could use?

n8bot, 2020-08-04:
I'm not 100% positive, but it seems like some/most/all of the placeholders are found in the source code via a line like this: ` m_placeholder_parser.set("initial_extruder", initial_extruder_id);` where you could search simply for `m_placeholder_parser.set`, with VSCode or Visual Studio, to find all the instances of those calls.

supermerill, 2020-08-04:
all variables are the settings (you can get them with the command superslicer.exe --help-fff) and somes other extra, that should be written in the tooltip.

Exerqtor, 2020-08-04:
Uuuhm ok, i'm not a programmer soo even though what both of guys just said is correct i have no idea of how to execute what you described and get to the info. Information like that should been much easier that access for the user imo (not critique aimed at you Merill). Pretty sure the majority of the userbase are bellow the knowlege threshold to obtain the information if this is the only way lol.

supermerill, 2020-08-04:
yeah, i know.
Ideally, documentation, tutorials, videos and wiki should exist...

Currently, i'm using mainly the tooltips and some wiki pages here for some specific things. So much to do.