supermerill/SuperSlicer #378: Use-case for thin_walls_merge

zbrozek, 2020-08-01:
I noticed in the tooltip that this feature was on the chopping block because it was not clear what it was good for. I use it!

I often design-in my own support structures. Disabling thin_wall_merge allows those supports to be very-weakly connected to the main structure, making them easy to remove. This is very useful and makes for very little cleanup.

supermerill, 2020-08-01:
thank you for the feedback!

robthide37, 2020-08-02:
> I noticed in the tooltip that this feature was on the chopping block because it was not clear what it was good for. I use it!
>
> I often design-in my own support structures. Disabling thin_wall_merge [...]

can you post example i am curious to try your method . thankyou in advance

zbrozek, 2020-08-02:
Sure, here's an example. It's a thumb splint which my wife designs and sells. The print needs some support to keep the arms reasonably stiff while printing up in Z, otherwise the nozzle drag deflects them and causes problems. It also modestly improves the arch up top. But it's also no good to have really well-connected support that we constantly have to cut and sand. These peel away by hand and take minimal finishing work.

The two-perimeter bit at the bottom is to trigger bridge flow and full connectedness to the main perimeters so that there's a build platform for the rest of the support wall.

[image]
[image]

robthide37, 2020-08-03:
Thankyou i have been debating builing a mmu2 or buying a palette but really have no problems with most supports and this looks like an interesting way to make delicate items easier to print