supermerill/SuperSlicer #379: print time with reprap and klipper

robthide37, 2020-08-02:
newest version any file you open if you select klipper or reprap acceleration doesn't change print time and if you set print speed to 200 or 150 or 120 it gives same time for all. I beleive iut is defaulting to a very low accel value for time estimation when not on marlin

robthide37, 2020-08-02:
you can set the accel in print settings but i don't want it to add anything to the gcode and feel like when set to klipper it should est time off the machine limits values entered on the printer settings vs needing to add it to the print settings as well

supermerill, 2020-08-02:
The slicer simulate marlin firmware. To have a better estimation for other firmware, you have to test & set the "time estimation compensation" in machine limits.

robthide37, 2020-08-03:
i get that what i am saying is that the acceleration doesn't work in the pinter config tab . so set acceleration to 10000 i use 6000 but just go higher for sake of proof. thenslice a file at 150mms then slice at 300mms the time might not change ir if anything will change by a few percent . If you change to marlin then it will behave correctly. the time estimation can't fix this problem becasue if you make it accurate for 200mms it still won't change when dropping to 150 mms. i print at 200 plus alot and when quality i will have 150 and 120 or 100 as my lowest . I set all speeds to the tested values just to make simple.

supermerill, 2020-08-03:
understood. I'll take a look.

robthide37, 2020-08-03:
thanks btw i asked in another thread how to donate more than the 3 or 5 a
month i would like to do a one lump sum ?

supermerill, 2020-08-03:
Problem is,the current github plateform only allow for month-based subscription.

1- Seems like github sponsor can support annual billing, so [if you set up your account for annual billing](https://github.com/account/upgrade?plan=free&plan_duration=year) and select the 2$ reward, you will pay 24$ in one go. Then, you should be able to [cancel](https://docs.github.com/en/github/setting-up-and-managing-billing-and-payments-on-github/downgrading-a-sponsorship#canceling-a-sponsorship) the support (but you'll be registered as support for the whole year).

2- If you prefer, I can create a 25$ tier, so you can select it and cancel, without changing the billing method.

note: I don't want to use paypal (or bitcoin), as it's possible to be scammed. With github, you are sure on who get the donation.

robthide37, 2020-08-07:
lmk if you will get the 24 in one shot ill do that if not make the 25 dollar tier and ill do that and then cancel and switch to the other one

closed 2020-10-22 by supermerill